chichengcn/gici-open #1: Is it possible to open-source the manual as well?

Hanbing59, 2023-07-04:
The repo is awesome! Great thanks to your team for sharing such an amazing project.

Maybe because I am a newbie in this field, I found it would be greater if the manual could be more and more detailed and specific. Of course, the work should not just be left to your team, it can be shared among all players who are interested in this project. 

Therefore, I am wondering if it is possible to upload and open-source the source code of the manual instead of merely the PDF file.

Again, thanks a lot for the fabulous work done by your team!

chichengcn, 2023-07-05:
Thanks for your affirmation. 

We will continuously optimize the documents to better serve a wider audience. But we think it is not necessary to open source the documentation. Please feel free to raise technical issues if you have any confusion about the algorithms or usages.

Hanbing59, 2023-07-05:
Thanks for letting me know what's in your mind.
Have a nice day!

closed 2023-07-05 by Hanbing59