chichengcn/gici-open #3: gici_main: command not found

sdjkjsdh, 2023-07-06:
Hi,
Thank for opensource this powerful software. And I am trying to run a demo but encounter some problems.
I think I have completed the compile successfully but still can not find the 'gici_main'.

The following is the log in the terminal after excuting the command 'make -j8'

bryan@bryan-virtual-machine:~/gici-open/build$ make -j8
[  0%] Building CXX object third_party/fast/CMakeFiles/fast.dir/src/fast_10.cpp.o
[  0%] Building C object third_party/rtklib/CMakeFiles/rtklib.dir/src/convgpx.c.o
[  1%] Building CXX object third_party/fast/CMakeFiles/fast.dir/src/fast_10_score.cpp.o
[  1%] Building CXX object third_party/fast/CMakeFiles/fast.dir/src/nonmax_3x3.cpp.o
[  2%] Building CXX object third_party/fast/CMakeFiles/fast.dir/src/faster_corner_10_sse.cpp.o
[  3%] Building C object third_party/rtklib/CMakeFiles/rtklib.dir/src/convkml.c.o
[  3%] Building C object third_party/rtklib/CMakeFiles/rtklib.dir/src/convrnx.c.o
[  4%] Building CXX object third_party/rpg_vikit/vikit_common/CMakeFiles/vikit_common.dir/src/vikit_cameras/camera_factory.cpp.o
[  5%] Building C object third_party/rtklib/CMakeFiles/rtklib.dir/src/datum.c.o
[  5%] Building C object third_party/rtklib/CMakeFiles/rtklib.dir/src/download.c.o
[  5%] Building CXX object third_party/rpg_vikit/vikit_common/CMakeFiles/vikit_common.dir/src/vikit_cameras/camera_geometry_base.cpp.o
[  6%] Building C object third_party/rtklib/CMakeFiles/rtklib.dir/src/ephemeris.c.o
[  7%] Building C object third_party/rtklib/CMakeFiles/rtklib.dir/src/geoid.c.o
[  7%] Building C object third_party/rtklib/CMakeFiles/rtklib.dir/src/gis.c.o
[  8%] Building C object third_party/rtklib/CMakeFiles/rtklib.dir/src/ionex.c.o
[  8%] Building C object third_party/rtklib/CMakeFiles/rtklib.dir/src/lambda.c.o
[  9%] Building C object third_party/rtklib/CMakeFiles/rtklib.dir/src/options.c.o
[  9%] Building C object third_party/rtklib/CMakeFiles/rtklib.dir/src/pntpos.c.o
[ 10%] Building C object third_party/rtklib/CMakeFiles/rtklib.dir/src/postpos.c.o
[ 10%] Building C object third_party/rtklib/CMakeFiles/rtklib.dir/src/ppp.c.o
[ 11%] Building C object third_party/rtklib/CMakeFiles/rtklib.dir/src/ppp_ar.c.o
[ 12%] Building C object third_party/rtklib/CMakeFiles/rtklib.dir/src/preceph.c.o
[ 12%] Building C object third_party/rtklib/CMakeFiles/rtklib.dir/src/rcvraw.c.o
[ 13%] Building C object third_party/rtklib/CMakeFiles/rtklib.dir/src/rinex.c.o
[ 13%] Building C object third_party/rtklib/CMakeFiles/rtklib.dir/src/rtcm.c.o
[ 14%] Building C object third_party/rtklib/CMakeFiles/rtklib.dir/src/rtcm2.c.o
[ 14%] Building C object third_party/rtklib/CMakeFiles/rtklib.dir/src/rtcm3.c.o
[ 15%] Building C object third_party/rtklib/CMakeFiles/rtklib.dir/src/rtcm3e.c.o
[ 16%] Building C object third_party/rtklib/CMakeFiles/rtklib.dir/src/rtkcmn.c.o
[ 16%] Linking CXX shared library ../../../third_party/fast/lib/libfast.so
[ 16%] Building C object third_party/rtklib/CMakeFiles/rtklib.dir/src/rtkpos.c.o
/usr/bin/c++ -fPIC -Wall -Werror -Wno-unknown-pragmas -fpermissive -mmmx -msse -msse -msse2 -msse3 -mssse3 -O3 -fsee -fomit-frame-pointer -fno-signed-zeros -fno-math-errno -funroll-loops  -shared -Wl,-soname,libfast.so -o ../../../third_party/fast/lib/libfast.so CMakeFiles/fast.dir/src/fast_10.cpp.o CMakeFiles/fast.dir/src/fast_10_score.cpp.o CMakeFiles/fast.dir/src/nonmax_3x3.cpp.o CMakeFiles/fast.dir/src/faster_corner_10_sse.cpp.o 
[ 16%] Built target fast
[ 17%] Building C object third_party/rtklib/CMakeFiles/rtklib.dir/src/rtksvr.c.o
[ 17%] Building C object third_party/rtklib/CMakeFiles/rtklib.dir/src/sbas.c.o
[ 18%] Building C object third_party/rtklib/CMakeFiles/rtklib.dir/src/solution.c.o
[ 18%] Building C object third_party/rtklib/CMakeFiles/rtklib.dir/src/stream.c.o
[ 19%] Building C object third_party/rtklib/CMakeFiles/rtklib.dir/src/streamsvr.c.o
[ 19%] Building C object third_party/rtklib/CMakeFiles/rtklib.dir/src/tides.c.o
[ 20%] Building C object third_party/rtklib/CMakeFiles/rtklib.dir/src/tle.c.o
[ 21%] Building C object third_party/rtklib/CMakeFiles/rtklib.dir/src/rcv/binex.c.o
[ 21%] Building C object third_party/rtklib/CMakeFiles/rtklib.dir/src/rcv/crescent.c.o
[ 22%] Building CXX object third_party/rpg_vikit/vikit_common/CMakeFiles/vikit_common.dir/src/vikit_cameras/camera_yaml_serialization.cpp.o
[ 23%] Building C object third_party/rtklib/CMakeFiles/rtklib.dir/src/rcv/javad.c.o
[ 23%] Building C object third_party/rtklib/CMakeFiles/rtklib.dir/src/rcv/novatel.c.o
[ 24%] Building C object third_party/rtklib/CMakeFiles/rtklib.dir/src/rcv/nvs.c.o
[ 24%] Building C object third_party/rtklib/CMakeFiles/rtklib.dir/src/rcv/rt17.c.o
[ 25%] Building C object third_party/rtklib/CMakeFiles/rtklib.dir/src/rcv/septentrio.c.o
[ 25%] Building C object third_party/rtklib/CMakeFiles/rtklib.dir/src/rcv/skytraq.c.o
[ 26%] Building C object third_party/rtklib/CMakeFiles/rtklib.dir/src/rcv/ss2.c.o
[ 27%] Building C object third_party/rtklib/CMakeFiles/rtklib.dir/src/rcv/ublox.c.o
[ 27%] Building CXX object third_party/rpg_vikit/vikit_common/CMakeFiles/vikit_common.dir/src/vikit_cameras/equidistant_fisheye_geometry.cpp.o
[ 28%] Building CXX object third_party/rpg_vikit/vikit_common/CMakeFiles/vikit_common.dir/src/vikit_cameras/equidistant_fisheye_projection.cpp.o
[ 28%] Building CXX object third_party/rpg_vikit/vikit_common/CMakeFiles/vikit_common.dir/src/vikit_cameras/ncamera.cpp.o
[ 29%] Building CXX object third_party/rpg_vikit/vikit_common/CMakeFiles/vikit_common.dir/src/vikit_cameras/ncamera_yaml_serialization.cpp.o
[ 29%] Building CXX object third_party/rpg_vikit/vikit_common/CMakeFiles/vikit_common.dir/src/vikit_cameras/omni_geometry.cpp.o
[ 29%] Linking C shared library ../../../third_party/rtklib/lib/librtklib.so
[ 29%] Built target rtklib
[ 30%] Building CXX object third_party/rpg_vikit/vikit_common/CMakeFiles/vikit_common.dir/src/vikit_cameras/omni_projection.cpp.o
[ 31%] Building CXX object third_party/rpg_vikit/vikit_common/CMakeFiles/vikit_common.dir/src/vikit_common/homography.cpp.o
[ 31%] Building CXX object third_party/rpg_vikit/vikit_common/CMakeFiles/vikit_common.dir/src/vikit_common/math_utils.cpp.o
[ 32%] Building CXX object third_party/rpg_vikit/vikit_common/CMakeFiles/vikit_common.dir/src/vikit_common/performance_monitor.cpp.o
[ 32%] Building CXX object third_party/rpg_vikit/vikit_common/CMakeFiles/vikit_common.dir/src/vikit_common/sample.cpp.o
[... 113 lines cut ...]
[ 96%] Building CXX object CMakeFiles/gici.dir/src/fusion/multisensor_estimating.cpp.o
[ 96%] Building CXX object CMakeFiles/gici.dir/src/fusion/rtk_imu_camera_rrr_estimator.cpp.o
[ 97%] Building CXX object CMakeFiles/gici.dir/src/fusion/rtk_imu_tc_estimator.cpp.o
[ 97%] Building CXX object CMakeFiles/gici.dir/src/fusion/spp_imu_camera_rrr_estimator.cpp.o
[ 98%] Building CXX object CMakeFiles/gici.dir/src/fusion/spp_imu_tc_estimator.cpp.o
[ 99%] Linking CXX shared library libgici.so
[ 99%] Built target gici
[ 99%] Building CXX object CMakeFiles/gici_main.dir/src/gici_main.cpp.o
[100%] Linking CXX executable gici_main
/usr/bin/ld: warning: libglog.so.0, needed by libgici.so, may conflict with libglog.so.1
/usr/bin/ld: CMakeFiles/gici_main.dir/src/gici_main.cpp.o: in function `YAML::Node YAML::Node::operator[]<char [8]>(char const (&) [8])':
gici_main.cpp:(.text._ZN4YAML4NodeixIA8_cEES0_RKT_[_ZN4YAML4NodeixIA8_cEES0_RKT_]+0x13e): undefined reference to `YAML::detail::node_data::convert_to_map(std::shared_ptr<YAML::detail::memory_holder> const&)'
/usr/bin/ld: libgici.so: undefined reference to `YAML::detail::node_data::empty_scalar[abi:cxx11]()'
/usr/bin/ld: ../third_party/rpg_vikit/vikit_common/lib/libvikit_common.so: undefined reference to `YAML::detail::node::m_amount'
/usr/bin/ld: ../third_party/rpg_vikit/vikit_common/lib/libvikit_common.so: undefined reference to `YAML::detail::node_data::push_back(YAML::detail::node&, std::shared_ptr<YAML::detail::memory_holder> const&)'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/gici_main.dir/build.make:157: gici_main] Error 1
make[1]: *** [CMakeFiles/Makefile2:165: CMakeFiles/gici_main.dir/all] Error 2
make: *** [Makefile:84: all] Error 2

/home/bryan/Pictures/Screenshot from 2023-07-06 23-00-59.png

[image]

Could you please help tell me what I have missed and how to solve this problem?

chichengcn, 2023-07-08:
Maybe you have installed the wrong version of yaml-cpp library. You should install it on https://github.com/jbeder/yaml-cpp, and then compile and install it. Before that, remember to uninstall your current-version yaml-cpp properly.

chichengcn, 2023-07-08:
Maybe you have installed the wrong version of yaml-cpp library. You should download it on https://github.com/jbeder/yaml-cpp, and then compile and install it. Before that, remember to uninstall your current version yaml-cpp properly.

sdjkjsdh, 2023-07-09:
> Maybe you have installed the wrong version of yaml-cpp library. You should download it on https://github.com/jbeder/yaml-cpp, and then compile and install it. Before that, remember to uninstall your current version yaml-cpp properly.

Thank you for the reply and the problem has been solved!

sdjkjsdh, 2023-07-09:
[image]

But why the GICI-LIB keeps watting for ephemeris and unable to load anything?

chichengcn, 2023-07-09:
Just ignore the INFO messages (Start with the term "I"), those are normal information. You should just care about the WARNING, ERROR, and FATAL messages.

GICI waits for sufficient ephemerides to start estimation, this process usually costs a few seconds. If the number of ephemerides is continually insufficient, you should check the "start_time" option in the RTCM formators. We stored the ephemerides in the RTCM format. Since it does not record the GPS week information, users should specify the record date to ensure correct RTCM decoding.

closed 2023-11-02 by chichengcn