chichengcn/gici-open #4: error: ‘LocalParameterization’ in namespace ‘ceres’ does not name a type

Haoyu0509, 2023-07-08:
Hi,
Thank for opensource this powerful software. And I am trying to run a demo but encounter some problems.
The following is the log in the terminal after excuting the command 'make -j8'

jap@jap-virtual-machine:~/gici/build$ make -j8
Scanning dependencies of target fast
Scanning dependencies of target rtklib
Scanning dependencies of target vikit_common
In file included from /home/jap/gici/include/gici/estimate/graph.h:55,
                 from /home/jap/gici/include/gici/estimate/estimator_types.h:30,
                 from /home/jap/gici/include/gici/stream/formator.h:19,
                 from /home/jap/gici/include/gici/stream/streaming.h:18,
                 from /home/jap/gici/include/gici/stream/data_integration.h:11,
                 from /home/jap/gici/src/stream/data_integration.cpp:9:
/home/jap/gici/include/gici/estimate/homogeneous_point_local_parameterization.h:57:40: error: expected class-name before ‘,’ token
   57 |     public ceres::LocalParameterization,
      |                                        ^
In file included from /home/jap/gici/include/gici/estimate/graph.h:56,
                 from /home/jap/gici/include/gici/estimate/estimator_types.h:30,
                 from /home/jap/gici/include/gici/stream/formator.h:19,
                 from /home/jap/gici/include/gici/stream/streaming.h:18,
                 from /home/jap/gici/include/gici/stream/data_integration.h:11,
                 from /home/jap/gici/src/stream/data_integration.cpp:9:
/home/jap/gici/include/gici/estimate/parameter_block.h:143:20: error: ‘LocalParameterization’ in namespace ‘ceres’ does not name a type
  143 |       const ceres::LocalParameterization* localParameterizationPtr)
      |                    ^~~~~~~~~~~~~~~~~~~~~
/home/jap/gici/include/gici/estimate/parameter_block.h:150:24: error: ‘LocalParameterization’ in namespace ‘ceres’ does not name a type
  150 |   virtual const ceres::LocalParameterization* localParameterizationPtr() const
      |                        ^~~~~~~~~~~~~~~~~~~~~
/home/jap/gici/include/gici/estimate/parameter_block.h:164:16: error: ‘LocalParameterization’ in namespace ‘ceres’ does not name a type
  164 |   const ceres::LocalParameterization* local_parameterization_ptr_;
      |                ^~~~~~~~~~~~~~~~~~~~~
/home/jap/gici/include/gici/estimate/parameter_block.h: In constructor ‘gici::ParameterBlock::ParameterBlock()’:
/home/jap/gici/include/gici/estimate/parameter_block.h:64:9: error: class ‘gici::ParameterBlock’ does not have any field named ‘local_parameterization_ptr_’
   64 |       , local_parameterization_ptr_(nullptr)
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/jap/gici/include/gici/estimate/parameter_block.h: In member function ‘virtual void gici::ParameterBlock::setLocalParameterizationPtr(const int*)’:
/home/jap/gici/include/gici/estimate/parameter_block.h:145:5: error: ‘local_parameterization_ptr_’ was not declared in this scope; did you mean ‘localParameterizationPtr’?
  145 |     local_parameterization_ptr_ = localParameterizationPtr;
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~
      |     localParameterizationPtr
In file included from /home/jap/gici/include/gici/estimate/graph.h:57,
                 from /home/jap/gici/include/gici/estimate/estimator_types.h:30,
                 from /home/jap/gici/include/gici/stream/formator.h:19,
                 from /home/jap/gici/include/gici/stream/streaming.h:18,
                 from /home/jap/gici/include/gici/stream/data_integration.h:11,
                 from /home/jap/gici/src/stream/data_integration.cpp:9:
/home/jap/gici/include/gici/estimate/pose_local_parameterization.h: At global scope:
/home/jap/gici/include/gici/estimate/pose_local_parameterization.h:62:40: error: expected class-name before ‘,’ token
   62 |     public ceres::LocalParameterization,
      |                                        ^
In file included from /home/jap/gici/include/gici/estimate/estimator_types.h:30,
                 from /home/jap/gici/include/gici/stream/formator.h:19,
                 from /home/jap/gici/include/gici/stream/streaming.h:18,
                 from /home/jap/gici/include/gici/stream/data_integration.h:11,
                 from /home/jap/gici/src/stream/data_integration.cpp:9:
/home/jap/gici/include/gici/estimate/graph.h:309:7: error: ‘ceres::LocalParameterization’ has not been declared
  309 |       ceres::LocalParameterization* local_parameterization);
      |       ^~~~~
/home/jap/gici/include/gici/estimate/graph.h:319:7: error: ‘ceres::LocalParameterization’ has not been declared
  319 |       ceres::LocalParameterization* local_parameterization)
      |       ^~~~~
In file included from /home/jap/gici/include/gici/estimate/graph.h:55,
                 from /home/jap/gici/include/gici/estimate/estimator_types.h:30,
                 from /home/jap/gici/include/gici/stream/formator.h:19,
                 from /home/jap/gici/src/utility/option.cpp:14:
/home/jap/gici/include/gici/estimate/homogeneous_point_local_parameterization.h:57:40: error: expected class-name before ‘,’ token
   57 |     public ceres::LocalParameterization,
      |                                        ^
In file included from /home/jap/gici/include/gici/estimate/graph.h:56,
[... 254 lines cut ...]
/home/jap/gici/include/gici/estimate/parameter_block.h:145:5: error: ‘local_parameterization_ptr_’ was not declared in this scope; did you mean ‘localParameterizationPtr’?
  145 |     local_parameterization_ptr_ = localParameterizationPtr;
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~
      |     localParameterizationPtr
In file included from /home/jap/gici/include/gici/estimate/graph.h:57,
                 from /home/jap/gici/include/gici/estimate/estimator_types.h:30,
                 from /home/jap/gici/include/gici/gnss/ambiguity_common.h:12,
                 from /home/jap/gici/src/gnss/ambiguity_common.cpp:9:
/home/jap/gici/include/gici/estimate/pose_local_parameterization.h: At global scope:
/home/jap/gici/include/gici/estimate/pose_local_parameterization.h:62:40: error: expected class-name before ‘,’ token
   62 |     public ceres::LocalParameterization,
      |                                        ^
In file included from /home/jap/gici/include/gici/estimate/estimator_types.h:30,
                 from /home/jap/gici/include/gici/gnss/ambiguity_common.h:12,
                 from /home/jap/gici/src/gnss/ambiguity_common.cpp:9:
/home/jap/gici/include/gici/estimate/graph.h:309:7: error: ‘ceres::LocalParameterization’ has not been declared
  309 |       ceres::LocalParameterization* local_parameterization);
      |       ^~~~~
/home/jap/gici/include/gici/estimate/graph.h:319:7: error: ‘ceres::LocalParameterization’ has not been declared
  319 |       ceres::LocalParameterization* local_parameterization)
      |       ^~~~~
[image]

I an sure I installed the Ceres-2.1.0 version.
Can you help me?

chichengcn, 2023-07-08:
This is because you downloaded the under-developing 2.2.0 version of Ceres on their github master repository. Ceres is replacing "LocalParameterization" with "Manifold" in the coming new version. 

You should download the source code, namely "the latest stable release", on http://ceres-solver.org/installation.html. Or you can directly download it at http://ceres-solver.org/ceres-solver-2.1.0.tar.gz.

Haoyu0509, 2023-07-08:
Thank you very much!

抄送: "吉安鹏 ***@***.******@***.***&gt;;

closed 2023-11-02 by chichengcn

Haoyu0509, 2023-11-10:
​
 抱歉，打扰了，池博，我在调试gici程序时遇到了头文件包含的问题。
  
  
​
 编辑
 

 然后我修改了c_cpp_properties.json和tasks.json文件
  
  
​
 编辑
 

  
  
​
 编辑
 

 但是出现了报错，显示找不到thirdparty下的头文件，您能帮我看看，指点一二吗？
  
​
 编辑
 
顺便贴一下launch.json文件
  
  
​
 编辑
 

 多谢池博！
 ​