chichengcn/gici-open #11: marginalization_error

ChanCody, 2023-09-18:
I'm getting this error randomly when I run `./build/gici_main option/pseudo_real_time_estimation_RTK_RRR.yaml`. I'm using the data 4.2 from gici-open-dataset. By randomly, I mean the "id" in the following log changes every time.  
The "id" can be 3250486229121942272, 3250486229112914688, 3250486229100626688, ...
```
I20230918 19:44:23.890939  9368 rtk_imu_camera_rrr_estimator.cpp:449] RTK/IMU/Camera RRR: Iterations: 6, Initial cost: 1.021e+02, Final cost: 1.005e+02, Sensor type: 3, Sat number:  8, GDOP: 4.3, Fix status: 3
I20230918 19:44:23.958727  9366 feature_handler.cpp:208] Select new keyframe by motivation: angle = 0.060711, distance = 0.589349
I20230918 19:44:23.959527  9368 gnss_estimator_base.h:633] Insufficient satellites! We need at least 8 satellites, but we only have 6!
I20230918 19:44:23.979460  9368 gnss_estimator_base.cpp:1584] Rejected pseudorange outlier with ambiguities at E15|7Q&7Q-E27|7Q&7Q: residual = 6.146627
I20230918 19:44:23.979605  9368 rtk_imu_camera_rrr_estimator.cpp:449] RTK/IMU/Camera RRR: Iterations: 6, Initial cost: 4.827e+02, Final cost: 1.352e+02, Sensor type: 1, Sat number:  8, GDOP: 4.3, Fix status: 3
F20230918 19:44:23.982164  9368 marginalization_error.cpp:606] Check failed: it != parameter_block_id_to_parameter_block_info_idx_.end() trying to marginalize out unconnected parameter block id = 3250486229121942272
*** Check failure stack trace: ***
    @     0x7f03c7c66e6e  google::LogMessage::Fail()
    @     0x7f03c7c66db6  google::LogMessage::SendToLog()
    @     0x7f03c7c665de  google::LogMessage::Flush()
    @     0x7f03c7c6a357  google::LogMessageFatal::~LogMessageFatal()
    @     0x7f03c872e748  gici::MarginalizationError::marginalizeOut()
    @     0x7f03c86f754c  gici::EstimatorBase::applyMarginalization()
    @     0x7f03c879685c  gici::RtkImuCameraRrrEstimator::gnssMarginalization()
    @     0x7f03c879bfac  gici::RtkImuCameraRrrEstimator::estimate()
    @     0x7f03c878af7e  gici::MultiSensorEstimating::processEstimator()
    @     0x7f03c878b690  gici::MultiSensorEstimating::runBackend()
    @     0x7f03c7983c80  (unknown)
    @     0x7f03c81156ba  start_thread
    @     0x7f03c73f251d  clone
    @              (nil)  (unknown)
Aborted
```
Is there something wrong with my configuration? I only changed the data path and start_time. Here  is the yaml file I'm using:
```
# GICI offline pseudo-real-time RTK/IMU/Camera RRR estimation
# You can uncomment the node configuration to publish related topics and observe estimation results (Build GICI in ROS wrapper).
stream:
  streamers:
    - streamer:
        tag: str_gnss_rov
        output_tags: [fmt_gnss_rov]
        type: file
        path: /home/chan/data/gici_data/4.2/gnss_rover.bin
    - streamer:
        tag: str_gnss_ref
        output_tags: [fmt_gnss_ref]
        type: file
        path: /home/chan/data/gici_data/4.2/gnss_reference.bin
    - streamer:
        tag: str_gnss_eph
        output_tags: [fmt_gnss_eph]
        type: file
        path: /home/chan/data/gici_data/4.2/gnss_ephemeris.bin
    - streamer:
        tag: str_dcb_file
        output_tags: [fmt_dcb_file]
        type: file
        path: /home/chan/develop/gici-open/option/CAS0MGXRAP_20221580000_01D_01D_DCB.BSX
        enable_time_tag: false
    - streamer:
        tag: str_imu
        output_tags: [fmt_imu]
        type: file
        path: /home/chan/data/gici_data/4.2/imu.bin
        loop_duration: 0.001
    - streamer:
        tag: str_camera
        output_tags: [fmt_camera]
        type: file
        path: /home/chan/data/gici_data/4.2/camera.bin
        buffer_length: 361472 # 752 * 480 + 512 
    - streamer:
        tag: str_rrr_solution_file
        input_tags: [fmt_rrr_solution_file]
        type: file
        path: /home/chan/data/gici_output/4.2.txt
[... 204 lines cut ...]
              cols: 1
              rows: 4
              data: [510.12091726559544, 509.55338413343736, 357.48596105253193, 240.00309579253124]
            distortion:
              type: radial-tangential  
              parameters:
                cols: 1
                rows: 4
                data: [-0.36472323465957157, 0.11530151924890532, 0.0006024586154990503, 0.0020184970860498133]
            mask: /home/chan/develop/gici-open/option/gici-mask.png
          T_B_C:
            cols: 4
            rows: 4
            data: [0.0134381210697455122,   -0.999901594987671172, -0.00402706384669154413,  -0.0229793554058591656,
                  0.999907681540912807,   0.0134460859659659704, -0.00195733688253543802,   0.0110786309679912626,
                  0.00201129251744842832, -0.00400038914436078377,     0.99998997574430859,    0.025008868367930974,
                                      0,                       0,                       0,                       1]

logging:
  enable: true
  min_log_level: 0
  log_to_stderr: true
  file_directory: /home/chan/data/gici_output/4.2.log
```
Appriciate for any help!

chichengcn, 2023-09-18:
Thanks for pointing it out. This is a BUG caused by our latest revision. We have fixed this BUG and uploaded the modified codes. 

There are no mistake on your configuration file. Just re-download the latest codes.

ChanCody, 2023-09-19:
I've pulled the latest codes, but the error still happens. Is this related to the " Insufficient satellites! We need at least 8 satellites, but we only have 6!" warning?
```
I20230919 09:24:32.619479 16317 rtk_imu_camera_rrr_estimator.cpp:449] RTK/IMU/Camera RRR: Iterations: 6, Initial cost: 9.936e+01, Final cost: 9.801e+01, Sensor type: 3, Sat number:  8, GDOP: 4.3, Fix status: 3
I20230919 09:24:32.690994 16315 feature_handler.cpp:208] Select new keyframe by motivation: angle = 0.0607363, distance = 0.569074
I20230919 09:24:32.691722 16317 gnss_estimator_base.h:633] Insufficient satellites! We need at least 8 satellites, but we only have 6!
I20230919 09:24:32.709928 16317 gnss_estimator_base.cpp:1584] Rejected pseudorange outlier with ambiguities at G03|5Q&5Q-G06|5Q&5Q: residual = -4.481940
I20230919 09:24:32.710001 16317 gnss_estimator_base.cpp:1584] Rejected pseudorange outlier with ambiguities at E15|7Q&7Q-E27|7Q&7Q: residual = 8.212878
I20230919 09:24:32.710157 16317 rtk_imu_camera_rrr_estimator.cpp:449] RTK/IMU/Camera RRR: Iterations: 6, Initial cost: 3.277e+02, Final cost: 1.285e+02, Sensor type: 1, Sat number:  8, GDOP: 4.3, Fix status: 3
F20230919 09:24:32.712699 16317 marginalization_error.cpp:606] Check failed: it != parameter_block_id_to_parameter_block_info_idx_.end() trying to marginalize out unconnected parameter block id = 3250486229122319104
*** Check failure stack trace: ***
    @     0x7fecfd668e6e  google::LogMessage::Fail()
    @     0x7fecfd668db6  google::LogMessage::SendToLog()
    @     0x7fecfd6685de  google::LogMessage::Flush()
    @     0x7fecfd66c357  google::LogMessageFatal::~LogMessageFatal()
    @     0x7fecfe130748  gici::MarginalizationError::marginalizeOut()
    @     0x7fecfe0f954c  gici::EstimatorBase::applyMarginalization()
    @     0x7fecfe19885c  gici::RtkImuCameraRrrEstimator::gnssMarginalization()
    @     0x7fecfe19dfac  gici::RtkImuCameraRrrEstimator::estimate()
    @     0x7fecfe18cf7e  gici::MultiSensorEstimating::processEstimator()
    @     0x7fecfe18d690  gici::MultiSensorEstimating::runBackend()
    @     0x7fecfd385c80  (unknown)
    @     0x7fecfdb176ba  start_thread
    @     0x7fecfcdf451d  clone
    @              (nil)  (unknown)
Aborted
```

147longlong, 2023-09-19:
Thanks for pointing it out. It's not related to the satellite number. 

There are also some BUG in our latest revision of non ROS interface. We will fix them a few days later and inform you.

Also, there may be no BUG or problem in our ROS interface, which has been tested many times in our different computers. You can build ROS interface and run it with the same configuration you mentioned earlier, refer to 2.2 in README.md

chichengcn, 2023-11-06:
We have tested the new codes on two different computers and the error did not happen. We suggest you clone the codes to a new folder and compile them, in case you did not merge the codes correctly.

closed 2024-01-03 by chichengcn