chichengcn/gici-open #12: 运行时星历缺失

ZiaoLiuS, 2023-09-20:
感谢作者您的开源，我在使用您的开源数据集1.1和1.2运行代码的过程中，程序一直显示等待星历。请问您是否遇到过此类问题。

ZiaoLiuS, 2023-09-20:
I20230919 23:08:00.352419 10806 data_integration.cpp:301] Waiting for ephemeris. We still have 35 satellites that do not have ephemeris. Total number of satellite is 44
I20230919 23:08:00.453202 10806 data_integration.cpp:301] Waiting for ephemeris. We still have 35 satellites that do not have ephemeris. Total number of satellite is 44
I20230919 23:08:00.551285 10806 data_integration.cpp:301] Waiting for ephemeris. We still have 35 satellites that do not have ephemeris. Total number of satellite is 44
I20230919 23:08:00.655683 10806 data_integration.cpp:301] Waiting for ephemeris. We still have 35 satellites that do not have ephemeris. Total number of satellite is 44

chichengcn, 2023-09-20:
请参考
https://github.com/chichengcn/gici-open-dataset/issues/4

ZiaoLiuS, 2023-09-20:
我按照数据集的真值文件给出了start_time，但似乎还是缺失星历。

ZiaoLiuS, 2023-09-20:
可以了，谢谢！

Ganglee98, 2023-09-23:
已经把注意把开始时间改成对应的时间。结果还是有等待星历的情况，请问还有哪里需要修改吗？data1.1时间不就是2023.3.20嘛

Ganglee98, 2023-09-23:
> 已经注意到把开始时间改成对应的时间。结果还是有等待星历的情况，请问还有哪里需要修改吗？data1.1时间不就是2023.3.20嘛

# GICI offline pseudo-real-time PPP estimation. 
stream:
  streamers:
  - streamer:
      tag: str_gnss_rov_file
      output_tags: [fmt_gnss_rov]
      type: file
      path: /home/.../Downloads/GICI/gici-open-master/data/1.1/gnss_rover.bin
  - streamer:
      tag: str_gnss_eph
      output_tags: [fmt_gnss_eph]
      type: file
      path: /home/.../Downloads/GICI/gici-open-master/data/1.1/gnss_ephemeris.bin
  - streamer:
      tag: str_gnss_ssr
      output_tags: [fmt_gnss_ssr]
      type: file
      path: /home/.../Downloads/GICI/gici-open-master/data/1.1/gnss_ssr.bin
  - streamer:
      tag: str_gnss_atx
      output_tags: [fmt_gnss_atx]
      type: file
      path: /home/.../Downloads/GICI/gici-open-master/option/igs14.atx
      enable_time_tag: false
  - streamer:
      tag: str_dcb_file
      output_tags: [fmt_dcb_file]
      type: file
      path: /home/.../Downloads/GICI/gici-open-master/option/CAS0MGXRAP_20221580000_01D_01D_DCB.BSX
      enable_time_tag: false
  - streamer:
      tag: str_solution_file
      input_tags: [fmt_solution_file]
      type: file
      path: /home/.../Downloads/GICI/gici-open-master/out_res/1.1/ppp_solution.txt
      enable_time_tag: false

  formators:
  - formator:
      io: input
      tag: fmt_gnss_rov
      type: gnss-raw
      sub_type: tersus
  - formator:
      io: input
      tag: fmt_gnss_eph
      type: gnss-rtcm-3
      start_time: 2023.03.20
  - formator:
      io: input
      tag: fmt_gnss_ssr
      type: gnss-rtcm-3
      start_time: 2023.03.20
  - formator:
      io: input
      tag: fmt_gnss_atx
      type: atx-file
  - formator:
      io: input
      tag: fmt_dcb_file
      type: dcb-file
  - formator:
      io: output
      tag: fmt_solution_file
      type: nmea 
      use_esa: true
      talker_id: GP

147longlong, 2023-09-23:
你好，data1.1的start_time就是2023.03.20。直观看来，您的配置文件没有问题，请核对以下几点：

1.以上内容中未含estimate，请在实际yaml中涵盖estimate。
2.确定是否出现的是星历等待问题？还是卫星数目不足问题？如果是后者，请确保gnss_ssr.bin在文件夹实际存在。
3.data2.1与1.1的start_time相同，可以将数据集替换成2.1再尝试。
4.如果以上都没有问题，请上传完整的yaml文件

Ganglee98, 2023-09-23:
> 你好，data1.1的start_time就是2023.03.20。直观看来，您的配置文件没有问题，请核对以下几点：
> 
> 1.以上内容中未含estimate，请在实际yaml中涵盖estimate。 2.确定是否出现的是星历等待问题？还是卫星数目不足问题？如果是后者，请确保gnss_ssr.bin在文件夹实际存在。 3.data2.1与1.1的start_time相同，可以将数据集替换成2.1再尝试。 4.如果以上都没有问题，请上传完整的yaml文件

十分感谢您的回复！下面我将完整上传yaml及运行遇到的问题，请您过目。

GICI offline pseudo-real-time PPP estimation.
stream:
streamers:
- streamer:
tag: str_gnss_rov_file
output_tags: [fmt_gnss_rov]
type: file
path: /home/.../Downloads/GICI/gici-open-master/data/4.1/gnss_rover.bin
- streamer:
tag: str_gnss_eph
output_tags: [fmt_gnss_eph]
type: file
path: /home/.../Downloads/GICI/gici-open-master/data/4.1/gnss_ephemeris.bin
- streamer:
tag: str_gnss_ssr
output_tags: [fmt_gnss_ssr]
type: file
path: /home/.../Downloads/GICI/gici-open-master/data/4.1/gnss_ssr.bin
- streamer:
tag: str_gnss_atx
output_tags: [fmt_gnss_atx]
type: file
path: /home/.../Downloads/GICI/gici-open-master/option/igs14.atx
enable_time_tag: false
- streamer:
tag: str_dcb_file
output_tags: [fmt_dcb_file]
type: file
path: /home/.../Downloads/GICI/gici-open-master/option/CAS0MGXRAP_20221580000_01D_01D_DCB.BSX
enable_time_tag: false
- streamer:
tag: str_solution_file
input_tags: [fmt_solution_file]
type: file
path: /home/liyuan/Downloads/GICI/gici-open-master/out_res/4.1/ppp_solution.txt
enable_time_tag: false

formators:
- formator:
io: input
tag: fmt_gnss_rov
type: gnss-raw
sub_type: tersus
- formator:
io: input
tag: fmt_gnss_eph
type: gnss-rtcm-3
start_time: 2023.05.21
- formator:
io: input
tag: fmt_gnss_ssr
type: gnss-rtcm-3
start_time: 2023.05.21
- formator:
io: input
tag: fmt_gnss_atx
type: atx-file
- formator:
io: input
tag: fmt_dcb_file
type: dcb-file
- formator:
io: output
tag: fmt_solution_file
[... 123 lines cut ...]
I20230923 16:55:06.884797 19244 data_integration.cpp:301] Waiting for ephemeris. We still have 37 satellites that do not have ephemeris. Total number of satellite is 44
I20230923 16:55:06.986246 19244 data_integration.cpp:301] Waiting for ephemeris. We still have 37 satellites that do not have ephemeris. Total number of satellite is 44
I20230923 16:55:07.091606 19244 data_integration.cpp:301] Waiting for ephemeris. We still have 37 satellites that do not have ephemeris. Total number of satellite is 44
I20230923 16:55:07.183110 19244 data_integration.cpp:301] Waiting for ephemeris. We still have 37 satellites that do not have ephemeris. Total number of satellite is 44
I20230923 16:55:07.284510 19244 data_integration.cpp:301] Waiting for ephemeris. We still have 37 satellites that do not have ephemeris. Total number of satellite is 44
I20230923 16:55:07.394009 19252 gnss_estimator_base.h:633] Insufficient satellites! We need at least 6 satellites, but we only have 0!
I20230923 16:55:07.490679 19252 gnss_estimator_base.h:633] Insufficient satellites! We need at least 6 satellites, but we only have 0!
I20230923 16:55:07.588677 19252 gnss_estimator_base.h:633] Insufficient satellites! We need at least 6 satellites, but we only have 0!
I20230923 16:55:07.689016 19252 gnss_estimator_base.h:633] Insufficient satellites! We need at least 6 satellites, but we only have 0!
I20230923 16:55:07.783695 19252 gnss_estimator_base.h:633] Insufficient satellites! We need at least 6 satellites, but we only have 0!
I20230923 16:55:07.886189 19252 gnss_estimator_base.h:633] Insufficient satellites! We need at least 6 satellites, but we only have 0!
I20230923 16:55:07.979715 19252 gnss_estimator_base.h:633] Insufficient satellites! We need at least 6 satellites, but we only have 0!
I20230923 16:55:08.090219 19252 gnss_estimator_base.h:633] Insufficient satellites! We need at least 6 satellites, but we only have 0!
I20230923 16:55:08.189184 19252 gnss_estimator_base.h:633] Insufficient satellites! We need at least 6 satellites, but we only have 0!
I20230923 16:55:08.288281 19252 gnss_estimator_base.h:633] Insufficient satellites! We need at least 6 satellites, but we only have 0!
I20230923 16:55:08.388388 19252 gnss_estimator_base.h:633] Insufficient satellites! We need at least 6 satellites, but we only have 0!
I20230923 16:55:08.491186 19252 gnss_estimator_base.h:633] Insufficient satellites! We need at least 6 satellites, but we only have 0!
I20230923 16:55:08.585942 19252 gnss_estimator_base.h:633] Insufficient satellites! We need at least 6 satellites, but we only have 0!
I20230923 16:55:08.686364 19252 gnss_estimator_base.h:633] Insufficient satellites! We need at least 6 satellites, but we only have 0!
I20230923 16:55:08.787806 19252 gnss_estimator_base.h:633] Insufficient satellites! We need at least 6 satellites, but we only have 0!
I20230923 16:55:08.884464 19252 gnss_estimator_base.h:633] Insufficient satellites! We need at least 6 satellites, but we only have 0!
I20230923 16:55:08.989281 19252 gnss_estimator_base.h:633] Insufficient satellites! We need at least 6 satellites, but we only have 0!
I20230923 16:55:09.095507 19252 gnss_estimator_base.h:633] Insufficient satellites! We need at least 6 satellites, but we only have 0!
I20230923 16:55:09.182811 19252 gnss_estimator_base.h:633] Insufficient satellites! We need at least 6 satellites, but we only have 0!
I20230923 16:55:09.287750 19252 gnss_estimator_base.h:633] Insufficient satellites! We need at least 6 satellites, but we only have 0!

Ganglee98, 2023-09-23:
> 你好，data1.1的start_time就是2023.03.20。直观看来，您的配置文件没有问题，请核对以下几点：
> 
> 1.以上内容中未含estimate，请在实际yaml中涵盖estimate。 2.确定是否出现的是星历等待问题？还是卫星数目不足问题？如果是后者，请确保gnss_ssr.bin在文件夹实际存在。 3.data2.1与1.1的start_time相同，可以将数据集替换成2.1再尝试。 4.如果以上都没有问题，请上传完整的yaml文件

Thanks a lot. I  use the datasets 4.1 which has the ssr data and then it works.
However, it still exists the following questions:

F20230923 17:37:30.726711 22140 marginalization_error.cpp:606] Check failed: it != parameter_block_id_to_parameter_block_info_idx_.end() trying to marginalize out unconnected parameter block id = 3534327355540504576
*** Check failure stack trace: ***
    @     0x7fa892a4d160  google::LogMessage::Fail()
    @     0x7fa892a4d0a1  google::LogMessage::SendToLog()
    @     0x7fa892a4c8be  google::LogMessage::Flush()
    @     0x7fa892a50564  google::LogMessageFatal::~LogMessageFatal()
    @     0x7fa892ef79b1  gici::MarginalizationError::marginalizeOut()
    @     0x7fa892eb7666  gici::EstimatorBase::applyMarginalization()
    @     0x7fa892dfc02c  gici::PppEstimator::marginalization()
    @     0x7fa892dfdfbf  gici::PppEstimator::estimate()
    @     0x7fa892f550df  gici::MultiSensorEstimating::processEstimator()
    @     0x7fa892f5591e  gici::MultiSensorEstimating::runBackend()
    @     0x7fa892923df4  (unknown)
    @     0x7fa892aeb609  start_thread
    @     0x7fa89275d133  clone
    @              (nil)  (unknown)
Aborted (core dumped)

Thank you again.

147longlong, 2023-09-23:
> Thanks a lot. I use the datasets 4.1 which has the ssr data and then it works. However, it still exists the following questions:
>
> F20230923 17:37:30.726711 22140 marginalization_error.cpp:606] Check [...]

If you use the non-ROS interface, it is a BUG of latest version. The author will fix it a few days later. Refer to https://github.com/chichengcn/gici-open/issues/11#issuecomment-1724975468

Ganglee98, 2023-09-23:
> > > 你好，data1.1的start_time就是2023.03.20。直观看来，您的配置文件没有问题，请核对以下几点：
> > > 1.以上内容中未含estimate，请在实际yaml中涵盖estimate。 2.确定是否出现的是星历等待问题？还是卫星数目不足问题？如果是后者，请确保gnss_ssr.bin在文件夹实际存在。 3.data2.1与1.1的start_time相同，可以将数据集替换成2.1再尝试。 4.如果以上都没有问题，请上传完整的yaml文件
> > 
> > 
> > Thanks a lot. I use the datasets 4.1 which has the ssr data and then it works. However, it still exists the following questions:
> > F20230923 17:37:30.726711 22140 marginalization_error.cpp:606] Check failed: it != parameter_block_id_to_parameter_block_info_idx_.end() trying to marginalize out unconnected parameter block id = 3534327355540504576 *** Check failure stack trace: *** @ 0x7fa892a4d160 google::LogMessage::Fail() @ 0x7fa892a4d0a1 google::LogMessage::SendToLog() @ 0x7fa892a4c8be google::LogMessage::Flush() @ 0x7fa892a50564 google::LogMessageFatal::~LogMessageFatal() @ 0x7fa892ef79b1 gici::MarginalizationError::marginalizeOut() @ 0x7fa892eb7666 gici::EstimatorBase::applyMarginalization() @ 0x7fa892dfc02c gici::PppEstimator::marginalization() @ 0x7fa892dfdfbf gici::PppEstimator::estimate() @ 0x7fa892f550df gici::MultiSensorEstimating::processEstimator() @ 0x7fa892f5591e gici::MultiSensorEstimating::runBackend() @ 0x7fa892923df4 (unknown) @ 0x7fa892aeb609 start_thread @ 0x7fa89275d133 clone @ (nil) (unknown) Aborted (core dumped)
> > Thank you again.
> 
> If you use the non-ROS interface, it is a BUG of the latest version. The author will fix it a few days later. Refer to [#11 (comment)](https://github.com/chichengcn/gici-open/issues/11#issuecomment-1724975468)

Thanks for your help.

closed 2023-11-08 by chichengcn