chichengcn/gici-open #13: does the gnss_imu_lc_estimator use GNSS Ambiguity Resolution?

wuyk1243, 2023-09-23:
Hi：）I want to use loosely coupled method to couple imu and GPS data. When I construct the data body of GNSS, need I provide variables values related to "GNSS Ambiguity Resolution", such as phase_bias and phase_center?  Or what value should I assign to these variables? since I only use the GPS sensor, I do not have those values yet.

chichengcn, 2023-09-23:
These values are used for PPP algorithm, you can ignore them if you are using RTK.

wuyk1243, 2023-09-25:
Thanks!🥹🥹🥹
So, if i have the GPS's position  velocity in ecef. How can I create a GnssSolutionStatus structure? (in https://github.com/chichengcn/gici-open/blob/bf4ab9d33f7e6158d5850c13c8f3aad64b37d7c6/include/gici/gnss/gnss_types.h#L230C2-L230C2 )
I don't know how to assign values to “id, covariance, status, num_satellites, differential_age”

chichengcn, 2023-10-02:
id is a unique number that used in estimator to distinguishes states. Just set as unique values for  different epochs.
covariance is a matrix describing the covariance of px, py, pz, vx, vy, vz. You can set them by the gnss algorithm output, or just by empirical.

The other variables you mentioned do not affect the LC estimators. You can set them casually.

closed 2023-11-08 by chichengcn