chichengcn/gici-open #14: logtostderr error

MayHarryWang, 2023-09-23:
Hello, thanks for providing such a powerful software.

I have some problem when running the example data. 

[image]

I have read the solution provided by https://blog.csdn.net/weixin_45432823/article/details/131542545 , but it seems not working.
I am not really sure what's the "sudo apt-get remove libgflags(tab补全)" in the post mean.
Can you provide me some suggestion?

Thank you very much!

chichengcn, 2023-09-23:
The immediate cause of this issue is that you have installed the glog library by apt-get, just as what we wrote in our readme file:

We suggest you install Glog from source code, rather than apt-get. Because installing from apt-get may make GICI fail to find the Glog library during compiling.

You should remove the glog library on your computer cleanly. And then re-install it from source code.

MayHarryWang, 2023-09-23:
Dear @chichengcn ,

Thanks for your reply, I just successfully run the example 1.1.
But there's an issue about cycle slip.
The first example is frequently cycle slipped for start to the end, as the figure shows. Does this issue also happen to you?
[image]

chichengcn, 2023-09-23:
The messages start with "I" are normal log messages. You do not need to care about them if there is no error happening.

MayHarryWang, 2023-09-24:
Dear @chichengcn 

Thanks for your patience. I want to make sure I have the same output in the RTK mode in this configuration.
Here's the start and end of the output file:
`$GPRMC,093604.400,A,3101.8264728,N,12126.6101636,E,0.06,0.00,200323,0.0,E,R,V*63
$GPGGA,093604.400,3101.8264728,N,12126.6101636,E,5,06,1.0,7.052,M,11.106,M,0.0,0000*7F
$GPESA,093604.400,+0.015,-0.012,+0.027,+0.000,+0.000,+0.000*7A
$GPESD,093604.400,0.704,0.726,1.761,0.086,0.125,0.249,0.000,0.000,0.000*7F
$GPRMC,093604.500,A,3101.8264750,N,12126.6101645,E,0.07,0.00,200323,0.0,E,R,V*68
$GPGGA,093604.500,3101.8264750,N,12126.6101645,E,5,06,1.0,7.045,M,11.106,M,0.0,0000*73
......
$GPESD,093912.600,0.019,0.019,0.045,0.089,0.108,0.088,0.000,0.000,0.000*7A
$GPRMC,093912.700,A,3101.8264808,N,12126.6106882,E,0.05,28.61,200323,0.0,E,R,V*5D
$GPGGA,093912.700,3101.8264808,N,12126.6106882,E,5,06,1.0,6.575,M,11.106,M,0.0,0000*7E
$GPESA,093912.700,+0.025,+0.008,+0.009,+0.000,+0.000,+0.000*73
$GPESD,093912.700,0.019,0.019,0.045,0.089,0.108,0.088,0.000,0.000,0.000*7B`

Is this output same as yours?

Besides, I have some questions about preparing my own data. I have collected some dataset in July and I try to apply these data in this powerful library.
1. Does this library support the GNSS ephemeris and observation file in traditional format? Such as RINEX 3.
If not, how can I convert the RINEX 3 file to the self-defined format? Such as gnss-rtcm-3 messages as the description in manual.

2. Does this library support tightly coupled PPP/INS?

Thank you very much!

chichengcn, 2023-10-08:
The result appears to be normal.

Currently we do not support RINEX 3, we will release the RINEX 3 support later, perhaps before the end of this year. The GNSS formats we are using are standardized and their definitions can be found online or in our source code. If you are in urgent need, you can add the necessary format conversion codes on your own.

We did not instantiate the PPP/INS TC , but it is easy for you to implement PPP/INS TC based on our object-oriented feature.

MayHarryWang, 2023-10-08:
Thanks for your kindly reply!

closed 2023-11-08 by chichengcn