chichengcn/gici-open #20: rank deficient when computing covariance of ppp_estimator

222233tommy, 2023-10-31:
Hi, Dr. chi, thanks for your amazing work. I try to test the PPP-AR and meet the rank deficient problem of the jacobian matrix. I guess the cause may be related to the single-differenced ambiguity leads to the rank = jacobian column -1. Could you please help me with that? Thanks a lot!

chichengcn, 2023-11-02:
We did not extensively test the PPP-AR module. Could you provide the configuration file and the number of datasets?

222233tommy, 2023-11-08:
Configuration file:

222233tommy, 2023-11-08:
There is another issue when I test the dual-frequency PPP with BDS. IFB for code 27 for system C has not been initialized!

chichengcn, 2023-11-10:
Thanks for pointing out. We have fixed this BUG and uploaded the codes. This BUG is caused by improper strategy of adding initial constraints for ambiguity and ionosphere parameters. Please note that phase bias product is demanded for PPP-AR algorithm, but we did not provide this product in our dataset. Forcely applying PPP-AR algorithm without phase bias corrections may cause large errors in solutions.

As for the IFB issue, code 27 (BDS B2bI) is not one of the base frequencies (B1I and B3I), however our dual-frequency PPP algorithm was designed for the base frequencies. So perhaps this BUG is foreseeable. But anyway, you can provide the configuration file and the number of dataset. We will focus on this issue in future development.

closed 2024-01-03 by chichengcn