chichengcn/gici-open #21: Inquiry PPP mode in ROS

Pirkaklo, 2023-11-01:
Thanks for your excellent work in multi-sensor fusion! Your expertise truly benefits our community!

I found that in your paper results, the PPP results have not been evaluated, but I am still curious about it.
I am using the dataset you provided [1.1, OpenSky area] to replicate your algorithm.
While I use 
```
rosrun gici_ros gici_ros_main  src/gici/option/ros_real_time_estimation_PPP.yaml
```
It would report
```
str_solution: At least one input tag or output tag should be specified!
```
And I found that in this YAML file, the output has been set as follows:
```
output_tags: [fmt_solution_file]
```
I try to modify it to 
```
output_tags: [str_solution, str_solution_odometry, str_solution_path, fmt_solution_file]
```
Then, it can work but without any result with 1.1 dataset, OpenSky area] dataset.
Here is the log:
```
I1101 15:44:51.313118 1447761 streamer.cpp:119] FileStreamer: Unable to load swap_interval. I am using default instead.
I1101 15:44:51.313148 1447761 streaming.cpp:104] str_gnss_atx: Unable to load buffer length! Using default instead.
I1101 15:44:51.313153 1447761 streaming.cpp:119] str_gnss_atx: Unable to load loop duration! Using default instead.
E1101 15:44:51.313205 1447761 streaming.cpp:139] Open streamer str_gnss_atx failed!
I1101 15:44:51.313210 1447761 streamer.cpp:119] FileStreamer: Unable to load swap_interval. Using default instead.
I1101 15:44:51.313213 1447761 streaming.cpp:104] str_dcb_file: Unable to load buffer length! Using default instead.
I1101 15:44:51.313215 1447761 streaming.cpp:119] str_dcb_file: Unable to load loop duration! Using default instead.
E1101 15:44:51.313223 1447761 streaming.cpp:139] Open streamer str_dcb_file failed!
I1101 15:44:51.313225 1447761 streamer.cpp:119] FileStreamer: Unable to load swap_interval. Using default instead.
I1101 15:44:51.313230 1447761 formator.cpp:921] NmeaFormator: Unable to load use_gga. Using default instead.
I1101 15:44:51.313230 1447761 formator.cpp:922] NmeaFormator: Unable to load use_rmc. Using default instead.
I1101 15:44:51.313231 1447761 formator.cpp:924] NmeaFormator: Unable to load use_esd. Using default instead.
I1101 15:44:51.313233 1447761 streaming.cpp:104] str_solution_file: Unable to load buffer length! Using default instead.
I1101 15:44:51.313236 1447761 streaming.cpp:119] str_solution_file: Unable to load loop duration! Using default instead.
I1101 15:44:51.313297 1447761 option.cpp:622] loadOptions<gici::EstimatorBaseOptions>: Unable to load force_initial_global_position. Using default instead.
I1101 15:44:51.313305 1447761 option.cpp:658] Unable to load initial_global_position. Using default instead.
I1101 15:44:51.313310 1447761 option.cpp:667] loadOptions<gici::GnssEstimatorBaseOptions>: Unable to load reject_one_outlier_once. Using default instead.
I1101 15:44:51.313313 1447761 option.cpp:670] loadOptions<gici::GnssEstimatorBaseOptions>: Unable to load max_doppler_error. Using default instead.
I1101 15:44:51.313315 1447761 option.cpp:671] loadOptions<gici::GnssEstimatorBaseOptions>: Unable to load good_observation_min_num_satellites. Using default instead.
I1101 15:44:51.313318 1447761 option.cpp:672] loadOptions<gici::GnssEstimatorBaseOptions>: Unable to load good_observation_max_gdop. Using default instead.
I1101 15:44:51.313318 1447761 option.cpp:673] loadOptions<gici::GnssEstimatorBaseOptions>: Unable to load good_observation_max_reject_ratio. Using default instead.
I1101 15:44:51.313320 1447761 option.cpp:674] loadOptions<gici::GnssEstimatorBaseOptions>: Unable to load reset_ambiguity_min_num_continuous_unfix. Using default instead.
I1101 15:44:51.313323 1447761 option.cpp:675] loadOptions<gici::GnssEstimatorBaseOptions>: Unable to load diverge_max_reject_ratio. Using default instead.
I1101 15:44:51.313324 1447761 option.cpp:676] loadOptions<gici::GnssEstimatorBaseOptions>: Unable to load diverge_min_num_continuous_reject. Using default instead.
I1101 15:44:51.313328 1447761 option.cpp:340] loadOptions<gici::GnssCommonOptions>: Unable to load min_num_satellite_redundancy. Using default instead.
I1101 15:44:51.313331 1447761 option.cpp:341] loadOptions<gici::GnssCommonOptions>: Unable to load max_gdop. Using default instead.
I1101 15:44:51.313333 1447761 option.cpp:342] loadOptions<gici::GnssCommonOptions>: Unable to load mw_slip_thres. Using default instead.
I1101 15:44:51.313334 1447761 option.cpp:343] loadOptions<gici::GnssCommonOptions>: Unable to load gf_slip_thres. Using default instead.
I1101 15:44:51.313336 1447761 option.cpp:344] loadOptions<gici::GnssCommonOptions>: Unable to load gf_sd_slip_thres. Using default instead.
I1101 15:44:51.313338 1447761 option.cpp:345] loadOptions<gici::GnssCommonOptions>: Unable to load period. Using default instead.
I1101 15:44:51.313341 1447761 option.cpp:395] Unable to load receiver_pco. Using default instead.
I1101 15:44:51.313344 1447761 option.cpp:404] loadOptions<gici::GnssErrorParameter>: Unable to load phase_error_factor. Using default instead.
I1101 15:44:51.313345 1447761 option.cpp:405] loadOptions<gici::GnssErrorParameter>: Unable to load doppler_error_factor. Using default instead.
I1101 15:44:51.313347 1447761 option.cpp:406] loadOptions<gici::GnssErrorParameter>: Unable to load ionosphere_broadcast_factor. Using default instead.
I1101 15:44:51.313349 1447761 option.cpp:407] loadOptions<gici::GnssErrorParameter>: Unable to load ionosphere_dual_frequency. Using default instead.
I1101 15:44:51.313350 1447761 option.cpp:408] loadOptions<gici::GnssErrorParameter>: Unable to load ionosphere_augment. Using default instead.
I1101 15:44:51.313351 1447761 option.cpp:409] loadOptions<gici::GnssErrorParameter>: Unable to load troposphere_model_factor. Using default instead.
I1101 15:44:51.313354 1447761 option.cpp:410] loadOptions<gici::GnssErrorParameter>: Unable to load troposphere_augment. Using default instead.
I1101 15:44:51.313354 1447761 option.cpp:411] loadOptions<gici::GnssErrorParameter>: Unable to load ephemeris_broadcast. Using default instead.
I1101 15:44:51.313356 1447761 option.cpp:412] loadOptions<gici::GnssErrorParameter>: Unable to load ephemeris_precise. Using default instead.
I1101 15:44:51.313359 1447761 option.cpp:413] loadOptions<gici::GnssErrorParameter>: Unable to load initial_position. Using default instead.
I1101 15:44:51.313360 1447761 option.cpp:414] loadOptions<gici::GnssErrorParameter>: Unable to load initial_velocity. Using default instead.
I1101 15:44:51.313361 1447761 option.cpp:415] loadOptions<gici::GnssErrorParameter>: Unable to load initial_clock. Using default instead.
I1101 15:44:51.313364 1447761 option.cpp:416] loadOptions<gici::GnssErrorParameter>: Unable to load initial_troposphere. Using default instead.
I1101 15:44:51.313365 1447761 option.cpp:417] loadOptions<gici::GnssErrorParameter>: Unable to load initial_ionosphere. Using default instead.
I1101 15:44:51.313367 1447761 option.cpp:418] loadOptions<gici::GnssErrorParameter>: Unable to load initial_ambiguity. Using default instead.
I1101 15:44:51.313369 1447761 option.cpp:419] loadOptions<gici::GnssErrorParameter>: Unable to load relative_troposphere. Using default instead.
I1101 15:44:51.313369 1447761 option.cpp:420] loadOptions<gici::GnssErrorParameter>: Unable to load relative_ionosphere. Using default instead.
I1101 15:44:51.313371 1447761 option.cpp:421] loadOptions<gici::GnssErrorParameter>: Unable to load relative_ambiguity. Using default instead.
I1101 15:44:51.313373 1447761 option.cpp:422] loadOptions<gici::GnssErrorParameter>: Unable to load relative_gps_ifcb. Using default instead.
I1101 15:44:51.313374 1447761 option.cpp:423] loadOptions<gici::GnssErrorParameter>: Unable to load relative_frequency. Using default instead.
I1101 15:44:51.313376 1447761 option.cpp:424] loadOptions<gici::GnssErrorParameter>: Unable to load residual_gps_ifcb. Using default instead.
I1101 15:44:51.313377 1447761 option.cpp:434] Unable to load system_error_ratio. Using default instead.
I1101 15:44:51.313378 1447761 option.cpp:446] Unable to load relative_position. Using default instead.
I1101 15:44:51.313380 1447761 option.cpp:458] Unable to load relative_velocity. Using default instead.
I1101 15:44:51.313689 1447761 ros_stream.cpp:49] Unable to load ROS topic queue size. Using default instead
I1101 15:44:51.314414 1447761 ros_stream.cpp:49] Unable to load ROS topic queue size. Using default instead
I1101 15:44:51.315337 1447761 ros_stream.cpp:49] Unable to load ROS topic queue size. Using default instead
Initialized 9 streamers, 3 formaters, and 1 estimator. Running...
```
I wonder if I made some mistakes, like not using precise clock/ephemeris files. Or this PPP part is still under development?

chichengcn, 2023-11-02:
There are two error messages on your log. Modify the file paths of str_gnss_atx and str_dcb_file.

Besides, did you convert our bin file to rosbag and play them after you run gici_ros_main? It seems that you forgot to do so.

Pirkaklo, 2023-11-03:
Thanks for your reply, I will check them.

closed 2024-01-03 by chichengcn