chichengcn/gici-open #23: 高斯牛顿法和LM

ZHAOFEGNSHUN, 2023-11-07:
[image]
池博 想问一下这里我们默认使用高斯牛顿法进行优化，那如果我换用LM方法，优化结果是不是会好一些呢？或者加个约束条件，对雅可比矩阵估计或可能存在数值进行约束，让它根据此进行自动切换，理论上是不是出来的精度会更好呢？请问这种想法是否可行？

chichengcn, 2023-11-07:
你找错地方了，我用的是Ceres-Solver做优化，优化方法可以通过.yaml文件配置，默认LM。你想对优化算法做文章的话，建议先到Ceres官网上学习一下优化算法。

ZHAOFEGNSHUN, 2023-11-09:
[image]
[image]
[image]
chi'lao'sh
池老师 我想问一下想着重看一下rtkimutc，这里trust_region_strategy_type默认是dogleg方法吗？如果我要是想修改成lm应该怎么改呀（我这次找对地方了吗呜呜呜呜呜）

chichengcn, 2023-11-09:
找对了。可以在src/utility/option.cpp下查看一下字符串和对应选项的映射关系。LM对应的配置文件上的字符串应该写“levenberg_marquardt”。

ZHAOFEGNSHUN, 2023-11-09:
好的好的感谢您！

closed 2024-01-03 by chichengcn