chichengcn/gici-open #24: GICI编译可以通过，但跑数据时报错

quangWHU, 2023-11-08:
报错信息如下：
I20231108 09:49:40.058630  3857 streamer.cpp:118] FileStreamer: Unable to load enable_time_tag. Using default instead.
I20231108 09:49:40.058671  3857 streamer.cpp:119] FileStreamer: Unable to load swap_interval. Using default instead.
I20231108 09:49:40.058686  3857 formator.cpp:314] loadStartTime: Unable to load start_time. Using default instead.
I20231108 09:49:40.061440  3857 streaming.cpp:104] str_gnss_rov: Unable to load buffer length! Using default instead.
I20231108 09:49:40.061455  3857 streaming.cpp:119] str_gnss_rov: Unable to load loop duration! Using default instead.
I20231108 09:49:40.061515  3857 streamer.cpp:118] FileStreamer: Unable to load enable_time_tag. Using default instead.
I20231108 09:49:40.061519  3857 streamer.cpp:119] FileStreamer: Unable to load swap_interval. Using default instead.
I20231108 09:49:40.064472  3857 streaming.cpp:104] str_gnss_ref: Unable to load buffer length! Using default instead.
I20231108 09:49:40.064483  3857 streaming.cpp:119] str_gnss_ref: Unable to load loop duration! Using default instead.
I20231108 09:49:40.064508  3857 streamer.cpp:118] FileStreamer: Unable to load enable_time_tag. Using default instead.
I20231108 09:49:40.064512  3857 streamer.cpp:119] FileStreamer: Unable to load swap_interval. Using default instead.
I20231108 09:49:40.067417  3857 streaming.cpp:104] str_gnss_eph: Unable to load buffer length! Using default instead.
I20231108 09:49:40.067430  3857 streaming.cpp:119] str_gnss_eph: Unable to load loop duration! Using default instead.
I20231108 09:49:40.067463  3857 streamer.cpp:119] FileStreamer: Unable to load swap_interval. Using default instead.
I20231108 09:49:40.067471  3857 streaming.cpp:104] str_dcb_file: Unable to load buffer length! Using default instead.
I20231108 09:49:40.067474  3857 streaming.cpp:119] str_dcb_file: Unable to load loop duration! Using default instead.
I20231108 09:49:40.067483  3857 streamer.cpp:118] FileStreamer: Unable to load enable_time_tag. Using default instead.
I20231108 09:49:40.067487  3857 streamer.cpp:119] FileStreamer: Unable to load swap_interval. Using default instead.
I20231108 09:49:40.067497  3857 streaming.cpp:104] str_imu: Unable to load buffer length! Using default instead.
I20231108 09:49:40.067519  3857 streamer.cpp:118] FileStreamer: Unable to load enable_time_tag. Using default instead.
I20231108 09:49:40.067523  3857 streamer.cpp:119] FileStreamer: Unable to load swap_interval. Using default instead.
I20231108 09:49:40.067530  3857 formator.cpp:749] ImagePackFormator: Unable to load step. Using default instead.
I20231108 09:49:40.067544  3857 streaming.cpp:119] str_camera: Unable to load loop duration! Using default instead.
I20231108 09:49:40.067557  3857 streamer.cpp:119] FileStreamer: Unable to load swap_interval. Using default instead.
I20231108 09:49:40.067562  3857 formator.cpp:921] NmeaFormator: Unable to load use_gga. Using default instead.
I20231108 09:49:40.067565  3857 formator.cpp:922] NmeaFormator: Unable to load use_rmc. Using default instead.
I20231108 09:49:40.067569  3857 formator.cpp:924] NmeaFormator: Unable to load use_esd. Using default instead.
I20231108 09:49:40.067574  3857 streaming.cpp:104] str_rrr_solution_file: Unable to load buffer length! Using default instead.
I20231108 09:49:40.067576  3857 streaming.cpp:119] str_rrr_solution_file: Unable to load loop duration! Using default instead.
I20231108 09:49:40.067603  3857 formator.cpp:921] NmeaFormator: Unable to load use_gga. Using default instead.
I20231108 09:49:40.067607  3857 formator.cpp:922] NmeaFormator: Unable to load use_rmc. Using default instead.
I20231108 09:49:40.067612  3857 streaming.cpp:104] str_solution_stream: Unable to load buffer length! Using default instead.
I20231108 09:49:40.067616  3857 streaming.cpp:119] str_solution_stream: Unable to load loop duration! Using default instead.
I20231108 09:49:40.067668  3857 option.cpp:622] loadOptions<gici::EstimatorBaseOptions>: Unable to load force_initial_global_position. Using default instead.
I20231108 09:49:40.067672  3857 option.cpp:623] loadOptions<gici::EstimatorBaseOptions>: Unable to load log_intermediate_data. Using default instead.
I20231108 09:49:40.067677  3857 option.cpp:624] loadOptions<gici::EstimatorBaseOptions>: Unable to load log_intermediate_data_directory. Using default instead.
I20231108 09:49:40.067683  3857 option.cpp:658] Unable to load initial_global_position. Using default instead.
I20231108 09:49:40.067693  3857 option.cpp:671] loadOptions<gici::GnssEstimatorBaseOptions>: Unable to load good_observation_min_num_satellites. Using default instead.
I20231108 09:49:40.067696  3857 option.cpp:672] loadOptions<gici::GnssEstimatorBaseOptions>: Unable to load good_observation_max_gdop. Using default instead.
I20231108 09:49:40.067699  3857 option.cpp:673] loadOptions<gici::GnssEstimatorBaseOptions>: Unable to load good_observation_max_reject_ratio. Using default instead.
I20231108 09:49:40.067703  3857 option.cpp:674] loadOptions<gici::GnssEstimatorBaseOptions>: Unable to load reset_ambiguity_min_num_continuous_unfix. Using default instead.
I20231108 09:49:40.067708  3857 option.cpp:675] loadOptions<gici::GnssEstimatorBaseOptions>: Unable to load diverge_max_reject_ratio. Using default instead.
I20231108 09:49:40.067711  3857 option.cpp:676] loadOptions<gici::GnssEstimatorBaseOptions>: Unable to load diverge_min_num_continuous_reject. Using default instead.
I20231108 09:49:40.067718  3857 option.cpp:340] loadOptions<gici::GnssCommonOptions>: Unable to load min_num_satellite_redundancy. Using default instead.
I20231108 09:49:40.067723  3857 option.cpp:341] loadOptions<gici::GnssCommonOptions>: Unable to load max_gdop. Using default instead.
I20231108 09:49:40.067726  3857 option.cpp:342] loadOptions<gici::GnssCommonOptions>: Unable to load mw_slip_thres. Using default instead.
I20231108 09:49:40.067729  3857 option.cpp:343] loadOptions<gici::GnssCommonOptions>: Unable to load gf_slip_thres. Using default instead.
I20231108 09:49:40.067732  3857 option.cpp:344] loadOptions<gici::GnssCommonOptions>: Unable to load gf_sd_slip_thres. Using default instead.
I20231108 09:49:40.067736  3857 option.cpp:345] loadOptions<gici::GnssCommonOptions>: Unable to load period. Using default instead.
I20231108 09:49:40.067744  3857 option.cpp:395] Unable to load receiver_pco. Using default instead.
I20231108 09:49:40.067754  3857 option.cpp:287] Unable to load gnss_error_parameter. Using default instead.
I20231108 09:49:40.067756  3857 option.cpp:693] loadOptions<gici::GnssLooseEstimatorBaseOptions>: Unable to load use_outlier_rejection. Using default instead.
I20231108 09:49:40.067759  3857 option.cpp:694] loadOptions<gici::GnssLooseEstimatorBaseOptions>: Unable to load max_position_error. Using default instead.
I20231108 09:49:40.067763  3857 option.cpp:695] loadOptions<gici::GnssLooseEstimatorBaseOptions>: Unable to load max_velocity_error. Using default instead.
I20231108 09:49:40.067766  3857 option.cpp:696] loadOptions<gici::GnssLooseEstimatorBaseOptions>: Unable to load diverge_min_num_continuous_reject. Using default instead.
I20231108 09:49:40.067772  3857 option.cpp:705] loadOptions<gici::ImuEstimatorBaseOptions>: Unable to load use_zupt. Using default instead.
I20231108 09:49:40.067775  3857 option.cpp:706] loadOptions<gici::ImuEstimatorBaseOptions>: Unable to load zupt_duration. Using default instead.
I20231108 09:49:40.067778  3857 option.cpp:707] loadOptions<gici::ImuEstimatorBaseOptions>: Unable to load zupt_max_acc_std. Using default instead.
I20231108 09:49:40.067782  3857 option.cpp:708] loadOptions<gici::ImuEstimatorBaseOptions>: Unable to load zupt_max_gyro_std. Using default instead.
I20231108 09:49:40.067785  3857 option.cpp:709] loadOptions<gici::ImuEstimatorBaseOptions>: Unable to load zupt_max_gyro_median. Using default instead.
I20231108 09:49:40.067788  3857 option.cpp:710] loadOptions<gici::ImuEstimatorBaseOptions>: Unable to load zupt_sigma_zero_velocity. Using default instead.
I20231108 09:49:40.067795  3857 option.cpp:514] loadOptions<gici::ImuParameters>: Unable to load a_max. Using default instead.
I20231108 09:49:40.067797  3857 option.cpp:515] loadOptions<gici::ImuParameters>: Unable to load g_max. Using default instead.
I20231108 09:49:40.067806  3857 option.cpp:522] loadOptions<gici::ImuParameters>: Unable to load rate. Using default instead.
[... cut ...]
    @     0x7f1f6b4d1ec0  vk::cameras::CameraGeometryBase::loadMask()
    @     0x7f1f6b4e1c33  vk::cameras::NCamera::loadFromYaml()
    @     0x7f1f6b6cdb7d  gici::option_tools::loadOptions<>()
    @     0x7f1f6b9c2dbf  gici::MultiSensorEstimating::MultiSensorEstimating()
    @     0x7f1f6b72c0db  gici::NodeHandle::NodeHandle()
    @     0x55a960487c53  main
    @     0x7f1f6af9b083  __libc_start_main
    @     0x55a9604880ce  _start
    @              (nil)  (unknown)
已放弃 (核心已转储)
解决办法：
仔细检查后发现mask文件路径没有添加，添加后可以正常运行

chichengcn, 2023-11-08:
LOG信息中，I开头的是INFO信息，无需关心。出问题一般检查W（WARN）、E（ERROR）、F（FATAL）开头的信息。

quangWHU, 2023-11-08:
问题已经解决，谢谢池老师

closed 2023-11-09 by chichengcn