chichengcn/gici-open #25: gici调试时遇到问题

Haoyu0509, 2023-11-08:
抱歉，打扰了，池博，我在调试gici程序时遇到了头文件包含的问题。
[image]
然后我修改了c_cpp_properties.json和tasks.json文件
[image]
[image]
但是出现了报错，显示找不到thirdparty下的头文件，您能帮我看看，指点一二吗？
[image]

顺便贴一下launch.json文件
[image]
多谢池博！

chichengcn, 2023-11-10:
不要使用vscode的debug窗口编译，其配置很麻烦，你直接使用我github上所述的cmake方式编译就行，debug的话需要把编译选项改为-DCMAKE_BUILD_TYPE=RelWithDebInfo

编译好后，仅需要修改launch.json就可以实现调试，下面是我的配置，供参考：

{
    // Use IntelliSense to learn about possible attributes.
    // Hover to view descriptions of existing attributes.
    // For more information, visit: https://go.microsoft.com/fwlink/?linkid=830387
    "version": "0.2.0",
    "configurations": [
        {
            "name": "g++ - Build and debug active file",
            "type": "cppdbg",
            "request": "launch",
            // "program": "/home/cc/linux/softwares/gici/build/gici_main",
            // "args": ["~/linux/softwares/gici/option/gic_replay.yaml"],
            "program": "${workspaceFolder}/ros_wrapper/devel/lib/gici_ros/gici_ros_main",
            "args": ["/home/cc/Work/Codes/gici/ros_wrapper/src/gici/option/ros_gic_tersus_replay.yaml"],
            // "program": "/home/cc/linux/softwares/gici/tools/ros/devel/lib/gici_tools/gici_files_to_rosbag",
            // "args": ["/home/cc/linux/softwares/gici/tools/ros/src/gici_tools/option/convert_rosbags.yaml"],
            // "program": "/home/cc/Work/Codes/gici/tools/evaluation/format_converters/build/imagepack_to_tum", 
            // "args": ["/home/cc/Work/Data/tmp/test1/log/image_20231023_0321.bin"],
            "stopAtEntry": false,
            "cwd": "${fileDirname}",
            "environment": [],
            "externalConsole": false,
            "MIMode": "gdb",
            "setupCommands": [
                {
                    "description": "Enable pretty-printing for gdb",
                    "text": "-enable-pretty-printing",
                    "ignoreFailures": true
                },
                {
                    "description": "Set Disassembly Flavor to Intel",
                    "text": "-gdb-set disassembly-flavor intel",
                    "ignoreFailures": true
                }
            ],
            "preLaunchTask": "C/C++: g++ build active file",
            "miDebuggerPath": "/usr/bin/gdb"
        }
    ]
}

Haoyu0509, 2023-11-10:
好的，多谢池博

closed 2024-01-03 by chichengcn