chichengcn/gici-open #26: Error: decode the 'gnss_rover.bin' to the rinex format

layfolk1999, 2023-12-22:
Hi author, I downloaded 'gici-open-dataset', I want to convert 'gnss_reference.bin' to normal obs, I changed the suffix '.bin' to 'rtcm3' and converted it to obs using rtklib's UI tool, rtkconv. but there is a problem when converting 'gnss_rover.bin' to obs as described above (the CRC24 never passes during decoding). I would like to ask if there is something else you are doing when generating 'gnss_rover.bin'.

layfolk1999, 2023-12-22:
Sorry, after checking the information, 'gnss_reference.bin' data is standard rtcm3 format, while 'gnss_rover.bin' data format needs to be changed to 'NovAtel OEM7', after the above adjustments, 'gnss_rover.bin' file can be converted to RINEX format normally.

[reference](https://blog.csdn.net/zhaolewen/article/details/133386720?spm=1001.2101.3001.6650.4&utm_medium=distribute.pc_relevant.none-task-blog-2%7Edefault%7ECTRLIST%7ERate-4-133386720-blog-131542545.235%5Ev39%5Epc_relevant_yljh&depth_1-utm_source=distribute.pc_relevant.none-task-blog-2%7Edefault%7ECTRLIST%7ERate-4-133386720-blog-131542545.235%5Ev39%5Epc_relevant_yljh&utm_relevant_index=5)

chichengcn, 2023-12-27:
ok.

closed 2023-12-27 by chichengcn