chichengcn/gici-open #28: 关于编译时遇到的问题

ZoYooJy, 2024-01-02:
```
[ 35%] Linking CXX shared library ../../../../third_party/rpg_vikit/vikit_common/lib/libvikit_common.so
/usr/bin/ld: /usr/local/lib/libyaml-cpp.a(scanscalar.cpp.o): relocation R_X86_64_PC32 against symbol `_ZNK4YAML5RegEx14MatchUncheckedINS_16StreamCharSourceEEEiRKT_' can not be used when making a shared object; recompile with -fPIC
/usr/bin/ld: final link failed: Bad value
collect2: error: ld returned 1 exit status
third_party/rpg_vikit/vikit_common/CMakeFiles/vikit_common.dir/build.make:344: recipe for target '../third_party/rpg_vikit/vikit_common/lib/libvikit_common.so' failed
make[2]: *** [../third_party/rpg_vikit/vikit_common/lib/libvikit_common.so] Error 1
CMakeFiles/Makefile2:267: recipe for target 'third_party/rpg_vikit/vikit_common/CMakeFiles/vikit_common.dir/all' failed
make[1]: *** [third_party/rpg_vikit/vikit_common/CMakeFiles/vikit_common.dir/all] Error 2
Makefile:102: recipe for target 'all' failed
make: *** [all] Error 2
```
编译时遇到的与第三方库rpg_vikit有关的问题，请问如何解决呢？谢谢

ZoYooJy, 2024-01-03:
已解决，是动态、静态库的链接问题。编译yaml-cpp库时需要开启-fPIC，即：cmake - DCMAKE_POSITION_INDEPENDENT_CODE=ON ..  make  sudo make install

chichengcn, 2024-01-03:
感谢分享。

closed 2024-01-03 by chichengcn