chichengcn/gici-open #29: Data conversion problem🥹

layfolk1999, 2024-01-03:
How to convert the IMU data(imu.bin) to the format(.imr) recognized by IE software? I did not find the corresponding "IMU Profiles" in IE software, and then customize its format, enter the information, the result can not be turned out. I would like to ask the author or everyone how to solve this problem, thank you！🫡

layfolk1999, 2024-01-03:
[image]

layfolk1999, 2024-01-03:
[image]

chichengcn, 2024-01-03:
We have provided the source code for bin to imr conversion. See https://github.com/chichengcn/gici-open/tree/master/tools/evaluation/format_converters/src/imupack_to_imr.cpp

layfolk1999, 2024-01-03:
thanks for your reply！😁

closed 2024-01-13 by layfolk1999