chichengcn/gici-open #31: 数据格式转换：bin to rosbag

ZoYooJy, 2024-01-05:
描述：通过gici_files_to_rosbag.cpp将dataset中的bin文件转换为rosbag，并以ros的形式运行代码。修改convert_rosbags.yaml中5个输入路径(bin), 5个输出路径(rosbag), 2个start_time(与goundtruth中的时间对应)
问题：只有dataset 1.1能够转换成功并正常运行，其余的output均无数据，且rviz中表现为marker status:error
[image]
[image]

ZoYooJy, 2024-01-05:
请问我是忽略了某些地方导致的错误么？谢谢！！！

chichengcn, 2024-01-05:
时间要与gici-open-dataset readme中的表格相对应，即

[image]

ZoYooJy, 2024-01-05:
好的，感谢回复！！！

closed 2024-02-27 by chichengcn