chichengcn/gici-open #33: The format of SSR correction

MayHarryWang, 2024-01-23:
Dear Dr. Chi,

Thanks for providing the powerful software.

I have one question when I'm reading the manual and code.
If I want to use the SSR corrected ephemeris feature, is the message format same as the Table 2.51 to 2.56 in the manual (without ROS)?
If so, what is the "std_msgs/Header" in the "Type" category mean?

On the other hand, I have a Septentrio receiver to collect the SSR augmentation data.
Is it possible to format the augmentation data to the same format as yours?
Actually, I want to format my SSR data and put in your software, but the data was not collected by using GICI.

Also, where can I find the SSR message decoders/functions?
If I can find the decoder, I think it's possible to format my data to fit the decoders.

Thank you very much.

Best regards.

chichengcn, 2024-02-27:
We support RTCM3 format SSR message as input for the non-ROS interface. The gnss_ssr.bin files contained in our datasets are in such format. These messages were collected from IGS CAS real-time SSR stream during experiment. 

As for the decoder, you can refer to the class RTCM3Formator in https://github.com/chichengcn/gici-open/blob/master/src/stream/formator.cpp.

MayHarryWang, 2024-02-27:
Thanks for your kindly reply!

I have one last question, do you also collect the ephemeris in RTCM format during the experiment?
If not, which tool did you convert the RINEX format to RTCM format?
I wonder if there's any tool that can help the format conversion.

Thanks you very much.

Best regards.

chichengcn, 2024-02-28:
gnss_reference.bin

chichengcn, 2024-02-28:
gnss_ephemeris.bin

closed 2024-03-16 by chichengcn