chichengcn/gici-open #37: Marginalization_error occurred when running dataset 5.2 using SPP_RRR

Fan-Yiheng, 2024-04-09:
Thanks to the author's selfless open source, I ran the **dataset 5.2 using SPP_RRR** through non-ROS and ROS ways，and both encountered a similar problem to #11. I have run it using **RTK_RRR and it works well, so I wonder if SPP_RRR method still have a similar problem.** It occurs around 850s in dataset 5.2 , here are my bash outputs:

```
I0409 22:31:36.881067 23992 spp_imu_camera_rrr_estimator.cpp:353] SPP/IMU/Camera RRR: Iterations: 6, Initial cost: 1.446e+01, Final cost: 1.430e+01, Sensor type: 3, Sat number: 15, GDOP: 3.1
I0409 22:31:36.896167 23992 spp_imu_camera_rrr_estimator.cpp:353] SPP/IMU/Camera RRR: Iterations: 6, Initial cost: 1.927e+01, Final cost: 1.595e+01, Sensor type: 1, Sat number: 16, GDOP: 2.9
I0409 22:31:36.932507 23990 feature_handler.cpp:208] Select new keyframe by motivation: angle = 0.435539, distance = 0.818635
I0409 22:31:37.005779 23992 spp_imu_camera_rrr_estimator.cpp:353] SPP/IMU/Camera RRR: Iterations: 6, Initial cost: 1.493e+01, Final cost: 1.442e+01, Sensor type: 3, Sat number: 16, GDOP: 2.9
I0409 22:31:37.020643 23992 gnss_estimator_base.cpp:1422] Rejected pseudorange outlier at C08|2I: residual = -11.310482
I0409 22:31:37.020779 23992 spp_imu_camera_rrr_estimator.cpp:353] SPP/IMU/Camera RRR: Iterations: 6, Initial cost: 1.688e+01, Final cost: 1.321e+01, Sensor type: 1, Sat number: 16, GDOP: 2.9
F0409 22:31:37.024216 23992 marginalization_error.cpp:606] Check failed: it != parameter_block_id_to_parameter_block_info_idx_.end() trying to marginalize out unconnected parameter block id = 2097551529641558016
*** Check failure stack trace: ***
@     0x7fb8b08cd25a  google::LogMessage::Fail()
@     0x7fb8b08cd195  google::LogMessage::SendToLog()
@     0x7fb8b08ccac6  google::LogMessage::Flush()
@     0x7fb8b08d0546  google::LogMessageFatal::~LogMessageFatal()
@     0x7fb8b1da0391  gici::MarginalizationError::marginalizeOut()
@     0x7fb8b1d60106  gici::EstimatorBase::applyMarginalization()
@     0x7fb8b1e44be6  gici::SppImuCameraRrrEstimator::gnssMarginalization()
@     0x7fb8b1e48941  gici::SppImuCameraRrrEstimator::estimate()
@     0x7fb8b1dfe10f  gici::MultiSensorEstimating::processEstimator()
@     0x7fb8b1dfe91e  gici::MultiSensorEstimating::runBackend()
@     0x7fb8b07a7df4  (unknown)
@     0x7fb8b04a9609  start_thread
@     0x7fb8b05e3353  clone
@              (nil)  (unknown)
已放弃 (核心已转储)
```

chichengcn, 2024-04-19:
Yep, I think it is the similar problem. We will work on it and fix this bug later.

chichengcn, 2024-04-21:
The BUG has been fixed. See the latest updates.

Fan-Yiheng, 2024-04-21:
> The BUG has been fixed. See the latest updates.
Thanks for fixing it so quickly and good luck with your work!

closed 2024-04-21 by Fan-Yiheng