chichengcn/gici-open #38: 对数据集进行分析时提出的问题

AnthonyCR68, 2024-04-13:
1.我发现原始的ground_truth.txt文件中没有速度真值。想请问作者我应该如何获得速度参考真值？我想分析速度误差；
[image]
2.当我对1.1、1.2等开阔天空数据集进行RTK_TC、PPP_TC、SPP、PPP、RTK五种模式解算后，结合参考真值分析位置误差，发现除RTK_TC和RTK之外的三种模式的天向误差RMS都高达1~2m，想问您一下这是为什么？
[image]
图为ppp_tc，数据集1.1位置误差
感谢池博！！

AnthonyCR68, 2024-04-15:
> 我有个问题还想请教一下，请问你跑了自己的数据集嘛，结果如何呢？谢谢

不好意思，我没有跑过

chichengcn, 2024-04-19:
PPP的定位解跟真值确实不自洽。因为真值我们是用RTK+光纤INS算的，而RTK的坐标受基站参考坐标的影响，我们实验的时候交替使用了两类测站，一为我们自己架设的接收机作为基站，二为千寻的VRS。其中，我们自己架设的基站的坐标我们并没有严格标定，只是使用SPP解取平均获得的，所以与真实值有数米的差很正常；对于千寻的基站，我们用的是WGS84的节点，而PPP服务产品用的是ITRF14的框架，这两者之间也会存在差异。此外，由于每段数据的采集数据较短，因此，其定位解也不能得到充分收敛。

所以，对于这组城市数据集来说，运行PPP算法并非重点，只是作为demo，因为我们认为，目前PPP算法或者PPP-AR算法并不太能在城市环境中达到稳定的定位。而由于PPP-RTK算法并没有通用的数据流支持，我们也并没有开放相关代码。

我们开发PPP算法时，通过接入IGS测站的数据流进行算法处理，并与IGS的坐标进行了对比，以验证算法开发是正确的。所以，如果你想要验证PPP相关算法，可以使用IGS测站。如果要进行车载实验，可以在初始的时候让设备在开阔天空环境等待半小时左右，待PPP充分收敛后，再进行实验。

chichengcn, 2024-04-19:
至于速度真值，我们确实没有输出，不过好在我们输出的位姿真值是100Hz的，你可以使用前后历元作差除以时间的方式计算速度真值，这样的计算方式误差在5ms*acc以内，其实是基本满足需求的，车载实验的加速度较低，所以不用过于担心这样计算的真值不准。

AnthonyCR68, 2024-04-24:
> PPP的定位解跟真值确实不自洽。因为真值我们是用RTK+光纤INS算的，而RTK的坐标受基站参考坐标的影响，我们实验的时候交替使用了两类测站，一为我们自己架设的接收机作为基站，二为千寻的VRS。其中，我们自己架设的基站的坐标我们并没有严格标定，只是使用SPP解取平均获得的，所以与真实值有数米的差很正常；对于千寻的基站，我们用的是WGS84的节点，而PPP服务产品用的是ITRF14的框架，这两者之间 [...]

感谢解答，对于您说的真值文件中，参考站的坐标是由SPP解算而不是严格定位标定得到的，说明参考站的坐标是有误差的。那我在使用真值和解算值进行误差计算时，应该怎么消除这样的一个误差呢？有没有补偿方法？

chichengcn, 2024-04-24:
目前我们使用这组公开数据集进行PPP相关处理的时候，是对误差直接减去均值的，这样做其实也足够了，我们认为在城市环境中，PPP实际误差的波动量要远大于其均值。

我们后续可能会再采集一些静态基站，以及开阔天空长时间运动的数据，以支持PPP相关算法面向测绘标准的评估。

AnthonyCR68, 2024-04-24:
> 目前我们使用这组公开数据集进行PPP相关处理的时候，是对误差直接减去均值的，这样做其实也足够了，我们认为在城市环境中，PPP实际误差的波动量要远大于其均值。
> 
> 我们后续可能会再采集一些静态基站，以及开阔天空长时间运动的数据，以支持PPP相关算法面向测绘标准的评估。

您说的误差减去均值，我的理解是：例如我用一组数据集进行多次PPP解算，然后计算每次的误差，再对他们求误差平均值，最后使用单次PPP的误差减去误差平均值得到最后的误差值。想问问您我这样的理解是正确的吗？

chichengcn, 2024-04-24:
这样是可以的。我可以确定的是，5月21号采集的数据都是使用的千寻的VRS，也就是说偏差是一致的；而其他日期的数据可能存在交替使用不同测站。

closed 2024-05-10 by chichengcn

oldfish163, 2024-07-17:
> 我有个问题还想请教一下，请问你跑了自己的数据集嘛，结果如何呢？谢谢

你好，方便请教一下你们是如何测试自己数据的吗