chichengcn/gici-open #39: backend pending to fatal error

ZhiweiHuang5520, 2024-04-17:
Hi, when I run RTK-LC, I met this:

W20240417 15:59:51.344967 140736750155328 multisensor_estimating.cpp:764] Large backend pending: 177 measurements are waiting!
I20240417 15:59:51.514225 140736750155328 rtk_estimator.cpp:267] RTK: Iterations: 2, Initial cost: 4.261e+01, Final cost: 3.985e+01, Sat number: 28, GDOP: 1.5, Dif distance: 1 km, Fix status: 4
I20240417 15:59:51.953251 140736750155328 gnss_imu_initializer.cpp:139] Full filling dynamic window!
W20240417 15:59:51.953328 140736750155328 multisensor_estimating.cpp:764] Large backend pending: 182 measurements are waiting!
W20240417 15:59:51.953473 140736750155328 multisensor_estimating.cpp:764] Large backend pending: 181 measurements are waiting!
I20240417 15:59:52.053224 140736750155328 gnss_estimator_base.cpp:1620] Rejected pseudorange outlier with ambiguities at C38|2I&2I-C43|2I&2I: residual = 4.508661
I20240417 15:59:52.111382 140736750155328 rtk_estimator.cpp:267] RTK: Iterations: 2, Initial cost: 3.675e+01, Final cost: 3.622e+01, Sat number: 29, GDOP: 1.4, Dif distance: 1 km, Fix status: 4
I20240417 15:59:52.508863 140736750155328 gnss_imu_initializer.cpp:139] Full filling dynamic window!
W20240417 15:59:52.508963 140736750155328 multisensor_estimating.cpp:764] Large backend pending: 186 measurements are waiting!
I20240417 15:59:52.674533 140736750155328 rtk_estimator.cpp:267] RTK: Iterations: 2, Initial cost: 3.205e+01, Final cost: 3.198e+01, Sat number: 28, GDOP: 1.5, Dif distance: 1 km, Fix status: 4
I20240417 15:59:53.125164 140736750155328 gnss_imu_initializer.cpp:139] Full filling dynamic window!
W20240417 15:59:53.125275 140736750155328 multisensor_estimating.cpp:764] Large backend pending: 192 measurements are waiting!
I20240417 15:59:53.306525 140736750155328 rtk_estimator.cpp:267] RTK: Iterations: 2, Initial cost: 2.491e+01, Final cost: 2.468e+01, Sat number: 28, GDOP: 1.5, Dif distance: 1 km, Fix status: 4
I20240417 15:59:53.721874 140736750155328 gnss_imu_initializer.cpp:139] Full filling dynamic window!
W20240417 15:59:53.721939 140736750155328 multisensor_estimating.cpp:764] Large backend pending: 198 measurements are waiting!
I20240417 15:59:53.874707 140736750155328 rtk_estimator.cpp:267] RTK: Iterations: 2, Initial cost: 2.596e+01, Final cost: 2.587e+01, Sat number: 27, GDOP: 1.5, Dif distance: 1 km, Fix status: 4
F20240417 15:59:54.323800 140736699799104 rotation-quaternion-inl.h:82] Check failed: (squaredNorm()) <= (static_cast<Scalar>(1.0)) + (EPS<Scalar>::normalization_value()) (1.05698 vs. 1.0001) 
F20240417 15:59:54.323815 140736750155328 rotation-quaternion-inl.h:82] Check failed: (squaredNorm()) <= (static_cast<Scalar>(1.0)) + (EPS<Scalar>::normalization_value()) (1.0569 vs. 1.0001) F20240417 15:59:54.00000F323801 140736070673984 2024 rotation-quaternion-inl.h417 8215:59:54.323841 Check failed: 140736070673984(rotation-quaternion-inl.h:82] Check failed: (squaredNorm()) <= (static_cast<Scalar>(1.0)) + (EPS<Scalar>::normalization_value()) (1.0569 vs. 1.0001) 
F00000000 00:00:00.000000 140736079066688 logging.cc:1841] RAW: Check data_->num_chars_to_log_ > 0 && data_->message_text_[data_->num_chars_to_log_ - 1] == '\n' failed: 
    @     0x7ffff55d2f85  google::LogMessage::Fail()
    @     0x7ffff55d2f85  google::LogMessage::Fail()
    @     0x7ffff55d2f85  google::LogMessage::Fail()
    @     0x7ffff55d6bf6  google::LogMessageFatal::~LogMessageFatal()
    @     0x7ffff55d6bf6  google::LogMessageFatal::~LogMessageFatal()
    @     0x7ffff560d678  google::RawLog__(google::LogSeverity, char const*, int, char const*, ...)
    @     0x7ffff55d2615  google::LogMessage::SendToLog()
    @     0x7ffff6ff3edd  kindr::minimal::RotationQuaternionTemplate<double>::RotationQuaternionTemplate(Eigen::Quaternion<double, 0> const&)
    @     0x7ffff55d241e  google::LogMessage::Flush()
    @     0x7ffff6ff3edd  kindr::minimal::RotationQuaternionTemplate<double>::RotationQuaternionTemplate(Eigen::Quaternion<double, 0> const&)
    @     0x7ffff6ff4446  kindr::minimal::QuatTransformationTemplate<double>::QuatTransformationTemplate(Eigen::Matrix<double, 3, 1, 0, 3, 1> const&, Eigen::Quaternion<double, 0> const&)
    @     0x7ffff55d6bf1  google::LogMessageFatal::~LogMessageFatal()
    @     0x7ffff6ff29fb  gici::HMCError::EvaluateWithMinimalJacobians(double const* const*, double*, double**, double**) const
    @     0x7ffff6ff4446  kindr::minimal::QuatTransformationTemplate<double>::QuatTransformationTemplate(Eigen::Matrix<double, 3, 1, 0, 3, 1> const&, Eigen::Quaternion<double, 0> const&)
    @     0x7ffff6ff2918  gici::HMCError::Evaluate(double const* const*, double*, double**) const
    @     0x7ffff70ba265  gici::NHCError::EvaluateWithMinimalJacobians(double const* const*, double*, double**, double**) const
    @     0x7ffff6ff3edd  kindr::minimal::RotationQuaternionTemplate<double>::RotationQuaternionTemplate(Eigen::Quaternion<double, 0> const&)
    @     0x7ffff73ad161  ceres::internal::ResidualBlock::Evaluate(bool, double*, double*, double**, double*) const
    @     0x7ffff6ff4446  kindr::minimal::QuatTransformationTemplate<double>::QuatTransformationTemplate(Eigen::Matrix<double, 3, 1, 0, 3, 1> const&, Eigen::Quaternion<double, 0> const&)
    @     0x7ffff70ba182  gici::NHCError::Evaluate(double const* const*, double*, double**) const
    @     0x7ffff729a5b3  gici::PoseError::EvaluateWithMinimalJacobians(double const* const*, double*, double**, double**) const
    @     0x7ffff73ad161  ceres::internal::ResidualBlock::Evaluate(bool, double*, double*, double**, double*) const
    @     0x7ffff729a4d0  gici::PoseError::Evaluate(double const* const*, double*, double**) const
    @     0x7ffff73ad161  ceres::internal::ResidualBlock::Evaluate(bool, double*, double*, double**, double*) const
Signal: SIGABRT (Aborted)

chichengcn, 2024-04-19:
This issue is similar to issue https://github.com/chichengcn/gici-open/issues/27. 

Due to our use of simulated real-time playback, if the performance of your computer is insufficient, the workflow will be blocked, and the program will automatically downsample the data, resulting in degraded performance. If the blocking is severe, the program may diverge, reset, and even crash. You can reduce the playback rate to ensure that there are no references to "pending" in the log.

ZhiweiHuang5520, 2024-04-24:
However, I run it on a 11th I7 CPU, and I find this issue occur only when running with debug. Can you explain the performace diff between debug and release mode.

closed 2024-04-26 by ZhiweiHuang5520

chichengcn, 2024-05-10:
对性能要求不是很高，目前实验室同学们的电脑装上linux系统都能正常跑。但用虚拟机的话，都跑不了。