chichengcn/gici-open #42: ROS2 and RiNEX support

sauravuprety21, 2024-05-28:
Hi, I wanted to request ROS 2 - Humble support for this awesome library.

Would it be possible for you to upgrade the ros_wrapper functionality  to ROS 2. 

Alternatively, could you extend you datatool sets such that ROS2 bags and RINEX files can be converted to the ROS1 bag required by the library.

Thank you!

chichengcn, 2024-06-12:
Sorry, we have no experience with ROS2. Nevertheless, the ROS interface is just a simple wrapper for the GICI functions. I think it is not difficult to extend this to ROS2 by yourself.

Currently we have a coarsely developed RINEX interface. It is not stable enough to be released formally. We will upload these codes to a separated branch latter.

chichengcn, 2024-06-13:
Uploaded. See branch [unstable](https://github.com/chichengcn/gici-open/tree/unstable).

closed 2024-06-29 by chichengcn