chichengcn/gici-open #45: 使用gici-open跑自己的数据

HariwW, 2024-06-03:
根据作者提供的说明文档，我们在作者提供的数据集上成功地运行了代码。此外，我们还将相机、INS和GNSS接受器安装在了无人机上，同时架设了GNSS基站，从而采集自己数据。对于GNSS部分，我们得到了Novatel OEM4格式的.gnss文件，RINEX格式的.24o和.24p文件，如下图所示：

[image]

为了满足输入要求，我们使用了[https://github.com/MKlimenko/rinex2rtcm3](url)所提供的代码，将RINEX格式的.24o和.24p文件转为了gnss_reference.rtcm3文件和gnss_ephemeris.rtcm3，然后修改了pseudo_real_time_estimation_DGNSS.yaml中的相关参数，并且将enable_time_tag全部改为了false，但是却运行不出来正确的结果，如下图所示：

[image]

个人认为这个报错来源于数据格式之间的转换，从而造成了部分信息的缺失，请问作者能否提供一个可靠的RINEX2RTCM的工具，或者扩展gici-open对于RINEX的支持。

再次感谢作者对于gici-open的开发和开源！

chichengcn, 2024-06-12:
你这里可能有两方面的错误：

1. 在这种情况下，enable_time_tag并不适合被改为false，false的情况下会使得所有数据被瞬间读入，使时序紊乱。正确的做法是使用tools文件夹下的generate_replay_tag工具，对你的rtcm数据生成tag，然后按照仿实时的方式回放。
2. 你的转换工具可能并没有把所有的卫星星历都成功转换，同时，它应该也没有成功的把reference station坐标转换到rtcm中。

我们目前有支持rinex接口的代码，但由于只是粗略开发，所以不太稳定，我们近期会将这些代码上传到另外一个非稳定分支，你可以参考。

chichengcn, 2024-06-13:
Rinex支持代码已上传，见[unstable](https://github.com/chichengcn/gici-open/tree/unstable)分支。

AnthonyCR68, 2024-06-19:
> Rinex支持代码已上传，见[unstable](https://github.com/chichengcn/gici-open/tree/unstable)分支。
想问一下您Rinex文件要如何使用，配置文件里面要做什么样的更改？您说的不太稳定指的是哪方面的不稳定？

chichengcn, 2024-06-29:
先试用tool下的[generate_tag](https://github.com/chichengcn/gici-open/blob/master/tools/edit_binary/generate_replay_tag/src/generate_tag.cpp)对rinex文件生成对应的tag，而后按照正常的仿实时回放配置文件使用，其中formator的数据类型改为“rinex”。

不稳定主要是没有经过大量的测试，其中已知的问题是rinex转rosbag时的burst play问题，可以参考我在 issue #50 关于“Besides, one of the reasons for naming the "unstable" ... ”的回复。

ShangfengShi, 2024-07-06:
使用tool下的[generate_tag](https://github.com/chichengcn/gici-open/blob/master/tools/edit_binary/generate_replay_tag/src/generate_tag.cpp)对示例数据重新生成对应的tag（SPP），会出现卡死的情况？
[image]

closed 2024-07-13 by chichengcn