chichengcn/gici-open #52: 采集数据时，Ntrip服务不可用

treememory, 2024-07-08:
非常感谢您很出色的工作！
目前我想基于我们的data_strorage.yaml例子，采集自己的数据。但是实验中发现Ntrip服务不可用，在其他平台（https://github.com/hanoi404/ntrip）测试Ntrip服务是没问题的。然后我又基于RTKLIB 2.4.3 b34的STRSVR进行Ntrip服务检查，发现也是没有数据。请问你们是怎么解决这个问题，可以同时采集Cors服务数据的呢？
[image]

谢谢

chichengcn, 2024-07-13:
你应该是使用的VRS服务，通过Ntrip接收数据，与单站服务不同的是，VRS服务需要用户发送粗略位置，以生成用户位置附近的VRS基站，如果用户不发送位置，则服务器也不会给用户广播差分信息。

遗憾的是，目前gici并不支持给Ntrip基站发送用户位置，不过，rtklib是支持发送用户位置的，配置一下即可，可以用rtklib把数据转一下，把数据通过TCP给过来，具体流程如下：

首先，打开rtklib的strsvr，将input配置为NTRIP Client，将output配置为TCP Server。为了给服务器发送位置，需要点击Options，并配置Lat/Lon/Height，这个坐标表明了你希望在哪个位置附近生成VRS，由于一般而言，5 km以内都算短基线差分，因此，这个位置不需要太精细。

[image]

在上述界面点击开始后，界面应该会显示能正常接收数据流了，此时，所配置的TCP Server也会广播数据流。从gici的配置文件中配置TCP Client，即可接收该数据流：

`    
- streamer:
        tag: str_vrs
        type: tcp-client
        ip: xxx.xxx.xxx.xxx
        port: xxxx
        ...
`

其中，port为strsvr中配置TCP Server时填入的port，IP可以为两种形式：
1）strsvr计算机与gici计算机为同一台计算机，则可以通过本机TCP通信接收，IP地址为127.0.0.1；
2）strsvr计算机与gici计算机为不同计算机，则需要将两台计算机连接到一个局域网，然后查看strsvr计算机的IP地址，并将其填入上述ip中，该IP通常为192.168.x.xxx。

treememory, 2024-07-13:
感谢详细的回答，按照您的设置，已经能够正常的接收数据~
但我这种方式，好像并不能获得到我们定义的数据格式的数据。Ntrip接收到的数据，只是gnss_reference.bin，gnss_ephemeris.bin怎么获得呢？针对星历，目前是在硬件串口中获得广播星历。
针对gnss_reference.bin、gnss_ephemeris.bin、gnss_ssr.bin这些数据，您有什么推荐的Ntrip服务来进行实时获取吗？从而能够采集自己的数据进行测试

chichengcn, 2024-07-13:
如果你在硬件串口获取了星历数据，那么就不用再在其他地方获取了

如果你串口中的观测数据和星历数据是同一种格式，比如同为novatel格式，那么直接输出到一个formator中，设置为novatel解码，并在接入到算法时将这路formator的role设置为[rover, ephemeris]即可。

如果串口中的观测数据和星历数据是不同格式，比如前者为novatel，后者为rtcm3，则运行算法的时候把硬件串口数据流的streamer输出到两个formator，并分别按两种解码格式来配置，后面在算法部分分别制定两个formator输入的role为[rover]和[ephemeris]即可。

不过需要注意的是，在接收机数据类型解码器中，如novatel格式中，有些电文信息我们算法不支持（我们用的rtklib的解码，有些电文类型rtklib的代码不支持），所以在解码星历时，如果信息不全，则需要补充一下解码程序。

我们的gnss_reference.bin有的数据是通过我们自己架设的基站采集的，另外有些数据是从千寻服务按照我前面给出的数据转发方式采集的。gnss_ephemeris.bin和gnss_ssr.bin都是通过IGS的实时数据流采集的，你可以去IGS官网申请一个实时数据流的账号，账号的权限包括全球MGEX测站的观测值数据流、各分析中心的SSR数据流、以及用于AGNSS的星历数据流等。

closed 2024-07-21 by chichengcn