chichengcn/gici-open #56: 能否通过配置文件实现spp+imu松耦合的功能？

936384885xy, 2024-08-14:
池博士您好，我参考项目示例配置文件ros_real_time_estimation_LC.yaml和ros_real_time_estimation_SPP.yaml自己修改了一个spp_lc的配置，但是运行后只有spp的estimator正常工作了，gnss_imu的estimator没有启动，我是否需要修改代码以实现这个功能？还是我的配置文件哪里有错误？
我改的配置文件内容如下：
# GICI offline pseudo-real-time GNSS/IMU LC estimation with ROS topics as input and output
stream:
  streamers:
    - streamer:
        tag: str_ros_gnss_rov
        type: ros
        topic_name: /gici/gnss_rover
        io: input
        format: gnss_raw
        enable_observation: true
    - streamer:
        tag: str_ros_gnss_eph
        type: ros
        topic_name: /gici/gnss_ephemeris
        io: input
        format: gnss_raw
        enable_ephemeris: true
        enable_ionosphere_parameter: true
    - streamer:
        tag: str_dcb_file
        output_tags: [fmt_dcb_file]
        type: file
        path: /home/l1/gici-open/option/CAS0MGXRAP_20221580000_01D_01D_DCB.BSX
        enable_time_tag: false
    - streamer:
        tag: str_ros_imu
        type: ros
        topic_name: /gici/imu_raw
        io: input
        format: imu
    - streamer:
        tag: str_spp_solution_file
        input_tags: [fmt_spp_solution_file]
        type: file
        path: /home/l1/gici-open/output/spp_solution.txt
        enable_time_tag: false
    - streamer:
        tag: str_lc_solution_file
        input_tags: [fmt_lc_solution_file]
        type: file
        path: /home/l1/gici-open/output/lc_solution.txt
        enable_time_tag: false
    - streamer:
        tag: str_solution
        type: ros
        topic_name: solution
        queue_size: 5
        io: output
        format: pose_stamped
    - streamer:
        tag: str_solution_odometry
        type: ros
        topic_name: solution_odometry
        queue_size: 10
        io: output
        format: odometry
        subframe_id: Body
    - streamer:
        tag: str_solution_path
        type: ros
        topic_name: solution_path
        queue_size: 100
        io: output
        format: path

  formators:
    - formator:
        io: input
        tag: fmt_dcb_file
        type: dcb-file
    - formator:
        io: output
        tag: fmt_spp_solution_file
        type: nmea
        use_esa: true
        use_esd: true
        talker_id: GP
    - formator:
        io: output
        tag: fmt_lc_solution_file
        type: nmea
        use_esa: true
        talker_id: GP

  replay:
    enable: false

estimate:
  - estimator:
      tag: est_spp
      type: spp
      input_tags: [str_ros_gnss_rov, str_ros_gnss_eph, fmt_dcb_file]
      str_ros_gnss_rov_roles: [rover]
      str_ros_gnss_eph_roles: [ephemeris]
      fmt_dcb_file_roles: [code_bias]
      output_tags: [fmt_spp_solution_file]
      output_align_tag: str_ros_gnss_rov
      compute_covariance: false
      spp_options:
        estimate_velocity: false
      gnss_estimator_base_options:
        gnss_common:
          min_elevation: 12.0
          min_SNR: [35.0, 30.0]
        use_outlier_rejection: true
        max_pesudorange_error: 10.0
      estimator_base_options:
        max_iteration: 10
        num_threads: 2
        max_solver_time: 0.05
        verbose_output: true
        solver_type: dense_normal_cholesky
        trust_region_strategy_type: levenberg_marquardt

  - estimator:
      tag: est_gnss_imu_lc
      type: gnss_imu_lc
      input_tags: [str_ros_imu, est_spp]
      str_ros_imu_roles: [major]
      est_spp_roles: [position_and_velocity]
      output_tags:
        [
          str_solution,
          str_solution_odometry,
          str_solution_path,
          fmt_lc_solution_file,
        ]
      output_align_tag: str_ros_imu
      output_downsample_rate: 40
      gnss_imu_lc_options:
        max_window_length: 5
      gnss_loose_estimator_base_options:
      imu_estimator_base_options:
        imu_parameters:
          sigma_bg: 1.0e-3
          sigma_ba: 1.0e-2
          sigma_g_c: 2.6700e-04
          sigma_a_c: 0.0112
          sigma_gw_c: 2.0881e-06
          sigma_aw_c: 2.7361e-04
        car_motion: true
        body_to_imu_rotation: [-90.0, -90.0, 0.0]
        body_to_imu_rotation_std: 5.0
        car_motion_min_velocity: 3.0
        car_motion_max_anguler_velocity: 5.0
      estimator_base_options:
        max_iteration: 5
        num_threads: 4
        max_solver_time: 0.05
        verbose_output: true
        solver_type: sparse_normal_cholesky
        trust_region_strategy_type: dogleg
      gnss_imu_initializer_options:
        max_iteration: 30
        num_threads: 4
        max_solver_time: 0.5
        gnss_extrinsics: [-0.029, 0.354, -0.042]
        gnss_extrinsics_initial_std: [0.0, 0.0, 0.0]
        time_window_length_slow_motion: 0.05
        time_window_length_dynamic_motion: 0.5
        min_acceleration: 0.5

logging:
  enable: true
  min_log_level: 0
  log_to_stderr: true
  file_directory: /home/l1/gici-open/output

chichengcn, 2024-08-14:
1. SPP配置中的compute_covariance应该置为true，因为LC程序需要SPP的协方差以设置误差权重。
2. SPP配置中的estimate_velocity应该置为true，因为LC配置中的est_spp_roles被设置为了position_and_velocity。当然，也可以在estimate_velocity为false的情况下，把est_spp_roles设置为position，但还是建议保留速度估计值，它在组合导航中发挥的作用很大。

936384885xy, 2024-08-15:
感谢您的回答，按您说的方法设置后确实可以工作了。我用1.2这个数据集测试发现几乎全程都会有warning。
W0815 09:05:04.828541 16244 covariance_impl.cc:647] Jacobian matrix is rank deficient. Number of columns: 75 rank: 74
W0815 09:05:04.828553 16244 graph.cpp:862] Failed to compute covariance!
请问这个会有什么影响吗？

chichengcn, 2024-08-18:
这是一个BUG，已修复并更新代码。

closed 2024-08-20 by 936384885xy